Mange/rtl8192eu-linux-driver #140: USB adapter TP-Link TL WN823N not working with Kernel 5.3

TheJags, 2019-09-18:
I have tried installing these drivers from here, as steps described in Readme, quite a few times with Ubuntu Mainline Kernel 5.3 (few RCs and with final 5.3) from this page: https://kernel.ubuntu.com/~kernel-ppa/mainline/ 

but Network Manager fails to detect any wireless networks.

Though these drivers (Realtek-4.4.x branch) are working just fine with Ubuntu kernel 5.2.14. My system is Ubuntu Mate 19.04 64-bit, with USB adapter TP-Link TL WN823N.

Many thanks.

CGarces, 2019-09-18:
Hi.

I have done some changes at #141 
Can you please build it  with that change?

Maybe the easiest way is compile it from this branch.
https://github.com/CGarces/rtl8192eu-linux-driver/tree/CGarces-patch-1

TheJags, 2019-09-18:
@CGarces thanks for the quick reply.

First I removed removed the drivers with:
sudo dkms remove rtl8192eu/1.0 --all

Then installed the kernel 5.3, rebooted the PC, tried to install drivers from your 5.3 branch, and getting this error:

ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found

What should I do next. Thanks, Jags

Here are the detailed steps:

1, **git clone https://github.com/CGarces/rtl8192eu-linux-driver.git;**

2, cd rtl8192eu-linux-driver;

3, $ sudo dkms add .;

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.
$

4, $ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...

'make' all KVER=5.3.0-050300-generic...(bad exit status: 2)

ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found

Error! Bad return status for module build on kernel: 5.3.0-050300-generic (x86_64)

Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
$

CGarces, 2019-09-18:
Hi.

Thanks for testing.

Remove the folder and start again but selecting a specific branch

`git clone --single-branch --branch <branchname> <remote-repo>`

So step 1 should be:

`git clone --single-branch --branch CGarces-patch-1 https://github.com/CGarces/rtl8192eu-linux-driver.git;`

TheJags, 2019-09-18:
@CGarces Thanks again :)

Some progress but got some error, and it's stuck at "depmod..." for past 15+ minutes

Also, now before the next try, do I need to do any "cleanup"? Thanks alot.

Here are the steps:

:::  :::
$ git clone --single-branch --branch CGarces-patch-1 https://github.com/CGarces/rtl8192eu-linux-driver.git;

$ cd rtl8192eu-linux-driver;
$ sudo dkms add .;

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.3.0-050300-generic......................
cleaning build area...

DKMS: build completed.

8192eu.ko:
Running module version sanity check.

**modinfo: ERROR: missing module or filename.**
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.3.0-050300-generic/updates/dkms/

**depmod...**

:::  :::

CGarces, 2019-09-18:
I'm nor sure about the error.
Try 
```
sudo depmod -a
sudo dkms install rtl8192eu/1.0;
```

TheJags, 2019-09-18:
Okay I got this message:

````
$ sudo depmod -a

$ sudo dkms install rtl8192eu/1.0;

Module rtl8192eu/1.0 already installed on kernel 5.3.0-050300-generic/x86_64

$ 
````

CGarces, 2019-09-18:
I'm not sure about the commands, I don't have a linux computer right now.

```
sudo dkms remove rtl8192eu
sudo dkms install rtl8192eu/1.0;
```
just to be sure that the last version is loaded.

TheJags, 2019-09-18:
> I'm not sure about the commands, I don't have a linux computer right now.
> 
> ```
> sudo dkms remove rtl8192eu
> sudo dkms install rtl8192eu/1.0;
> ```
> 
> just to be sure that the last version is loaded.

Well, even before I read your message, I just rebooted the PC, and after reboot it got connected automatically right away.

Now when I tried the first command you mentioned, I get this error:

````
$ sudo dkms remove rtl8192eu

Error! Invalid number of parameters passed.

Usage: remove <module>/<module-version> --all
   or: remove <module>/<module-version> -k <kernel-version>
$ 

````

ipaleka, 2019-09-19:
> Hi.
> 
> I have done some changes at #141
> Can you please build it with that change?
> 
> Maybe the easiest way is compile it from this branch.
> https://github.com/CGarces/rtl8192eu-linux-driver/tree/CGarces-patch-1

Had the same issue with TP-Link TL-WN822N on 5.3.0 and it now works flawlessly after installing driver from this branch.

\o/

CGarces, 2019-09-23:
Hi @ipaleka 

PR #141 has been added to the repository so
https://github.com/Mange/rtl8192eu-linux-driver/
and 
https://github.com/CGarces/rtl8192eu-linux-driver/tree/CGarces-patch-1

Must work in the same way.

In any case issue is closed by PR #141

closed 2019-09-23 by CGarces

CGarces, 2020-09-14:
> Hi,
> 
> I have to install drivers of TL-WN 823N (version: 3.0) on Ubuntu 18.04 (Kernel: Linux 5.3.0-28-generic). I have searched the web-page of tp-link and also other forums but unable to get drivers and installation guide for mentioned version and OS.
> 
> Kindly help out in the said matter.
> 
> Thanks and Regards,

Please check README.md

halarhaleem, 2020-09-14:
> Please check README.md

Thankyou very much @CGarces , Its connected. 

Regards,

MrRa88it, 2021-09-30:
how to install  TL-WN823N driver on kali linux 2021?