Mange/rtl8192eu-linux-driver #143: Not working for TL-WN821N V6

dhillonr, 2019-10-03:
I ran all the steps successfully.

Received warning in step : 9 ->  sudo update-grub; sudo update-initramfs -u;

logs are : 

update-initramfs: Generating /boot/initrd.img-4.15.0-65-generic
W: Possible missing firmware /lib/firmware/i915/kbl_guc_ver9_14.bin for module i915
W: Possible missing firmware /lib/firmware/i915/bxt_guc_ver8_7.bin for module i915

After rebooting when i ran this :   
sudo lshw -c network;
It didn't list the required driver=8192eu

CGarces, 2019-10-03:
Hi 
i915 it a intel GPU device, not related with this driver.
Can you paste the id of your device? use `lusb`
Also you can see the loaded modules with `lsmod`

dhillonr, 2019-10-03:
Output of lsusb : 
Bus 002 Device 002: ID 8087:8000 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:8008 Intel Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 003 Device 006: ID 1a81:2004 Holtek Semiconductor, Inc. 
Bus 003 Device 005: ID 2357:0107  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

dhillonr, 2019-10-03:
lsmod output : 
8192eu               1085440  0

CGarces, 2019-10-03:
```
lsmod output :
8192eu 1085440 0
```
So, module is loaded.
And your device is supported.
https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/os_dep/linux/usb_intf.c#L192

Any error on dmesg?

dhillonr, 2019-10-03:
Their is something wrong with the device. Drivers are installed properly(Checked it with other device).
Will get it replaced from amazon.
Sorry for the inconvenience.

closed 2019-10-03 by dhillonr