Mange/rtl8192eu-linux-driver #145: Compatibility list

dAnjou, 2019-10-13:
Hi,

just wanted to ask whether it'd make sense to maintain a compatibility list? People could confirm working device/kernel combinations to save others some time. It took me for example a bit to find a driver that works.

Anyway, here's mine (just in case someone googles for it):

Device model | `uname -a`
------------ | -------------
TL-WN823N(EU) V3.0 | `Linux xv15 5.2.18-200.fc30.x86_64 #1 SMP Tue Oct 1 13:14:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

closed 2019-10-15 by CGarces

dAnjou, 2019-10-15:
Labeled question but closed without answer?

CGarces, 2019-10-15:
Hi @dAnjou 
Maintain a compatibility list it's impossible. Too many adaptors and too many kernel versions.