Mange/rtl8192eu-linux-driver #146: fedora 30 doesnt work with kernel 5.3

tester1969, 2019-10-22:
After upgrading to kernel 5.3 from 5.2, linux cannot load driver, it stays in loading stage forever. Going back to 5.2 works. Recompiled latest version from here. 
There is no error message on any log file or screen. 
Driver compiles ok.

SimonChatillon, 2019-10-30:
I have the same issue with Fedora 30, kernel 5.3.6. I see no wifi driver with `lshw -c network`, only ethernet. Also I get system errors each time I plug the dongle : "tainted module: 8192eu".

Let us know if any solution comes out

areal060781, 2019-10-30:
I have the same issue with kernel 5.3.7

tester1969, 2019-10-30:
OK tested with RHEL:
[root@oc3588317528 rtl8192eu-linux-driver-realtek-4.4.x]# modinfo 8192eu.ko |grep version
version:        v4.4.1_17696.20160509_BTCOEX20160412-0042
srcversion:     4B7E4909909179107BA2B63
vermagic:       5.3.8-1.el7.elrepo.x86_64 SMP mod_unload modversions 
[root@oc3588317528 rtl8192eu-linux-driver-realtek-4.4.x]# uname -a
Linux oc3588317528 5.3.8-1.el7.elrepo.x86_64 #1 SMP Tue Oct 29 08:30:10 EDT 2019 x86_64 x86_64 x86_64 GNU/Linux

This is with elrepo kernel 5.3.8 and it works. I haven't access till tomorrow to fedora box, i ll try to use the elrepo kernel there too. Anyone to try in his box the elrepo kernel?

CGarces, 2019-11-09:
Hi @tester1969 @areal060781 @SimonChatillon 
I have no idea about how fix this problem
¿Can you please test this branch?
https://github.com/CGarces/rtl8192eu-linux-driver/commits/realtek-5.x.x
It contains the driver v5.6.4_35685.20191108_COEX20171113-0047 from Realtek FTP plus all the patches from 4.4.x branch.

tester1969, 2019-11-09:
it works
[theodoore@localhost ~]$ modinfo 8192eu |grep version
version:        v5.6.4_35685.20191108_COEX20171113-0047
srcversion:     6294FFE9DE7182467505B5F
parm:           rtw_chip_version:int
[theodoore@localhost ~]$ uname -a
Linux localhost.localdomain 5.3.8-200.fc30.x86_64 #1 SMP Tue Oct 29 14:46:22 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
[theodoore@localhost ~]$ 
I upgraded however to 5.3.8

CGarces, 2019-11-19:
Closed by #149

closed 2019-11-19 by CGarces