Mange/rtl8192eu-linux-driver #147: Does not work on NVIDIA Jetson Nano

sargyrop, 2019-11-06:
I am trying to install the TP-Link WN-823N v2 usb dongle (Bus 001 Device 005: ID 2357:0109 from lsusb) in NVIDIA Jetson Nano (stock image, Ubuntu 18.04.3 LTS). I have followed the instructions in the README. The wireless networks appear correctly but I cannot connect: it keeps asking for the password (even though it is 100% correct). 

Also, using `nmcli dev wifi` I see that the signal strength is 94% and the rate is 2Mbit/s.

Some debugging output:

```
$ lsmod  
Module                  Size  Used by  
btrfs                1156495  0  
xor                     7364  1 btrfs  
raid6_pq               92714  1 btrfs  
bnep                   16562  2  
fuse                  103841  2  
zram                   26166  4  
cfg80211              589351  0  
overlay                48691  0  
nvgpu                1569917  35  
8192eu               1239728  0  
ip_tables              19441  0  
x_tables               28951  1 ip_tables  
```   

``` 
uname -a
Linux streamowl-desktop 4.9.140-tegra #1 SMP PREEMPT Sat Oct 19 15:54:06 PDT 2019 aarch64 aarch64 aarch64 GNU/Linux
```
```
sudo lshw -C network | grep 8192
       configuration: broadcast=yes driver=rtl8192eu multicast=yes wireless=unassociated
```

```
dmesg |grep 8192
[    2.351171] 8192eu: loading out-of-tree module taints kernel.
[    2.389850] RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
[    2.389855] RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
[    2.597139] usbcore: registered new interface driver rtl8192eu
```

CGarces, 2019-11-09:
I'm not sure if the problem with the password is related to a NetworkManager bug.

Thease read https://github.com/Mange/rtl8192eu-linux-driver/issues/46#issuecomment-325977795

And check also issue #57

sargyrop, 2019-11-14:
The _wifi.scan-rand-mac-address=no_ line is already there. Also, the rtl8xxxu driver is already blacklisted.

CGarces, 2019-11-14:
Can you please test this branch?
https://github.com/CGarces/rtl8192eu-linux-driver/tree/realtek-5.x.x

sargyrop, 2019-11-14:
The Makefile does not have an `CONFIG_PLATFORM_ARM_AARCH64 = y` entry. Shall I edit it manually?

sargyrop, 2019-11-14:
Using the above entry ```CONFIG_PLATFORM_ARM_AARCH64 = y``` the problem is still there.
However, in ```dmesg```, now I see the following:

``` 
 RTL871X: module init start
[    2.484791] RTL871X: rtl8192eu v4.4.1_17696.20160509_BTCOEX20160412-0042
[    2.484794] RTL871X: build time: Nov 14 2019 10:15:58
[    2.484796] RTL871X: rtl8192eu BT-Coex version = BTCOEX20160412-0042
[    2.679486] RTL871X: hal_com_config_channel_plan chplan:0x21
[    2.684468] RTL871X: Wrong rate section: (CCK,2T)
...
                      RTL871X: Wrong rate section: (CCK,2T)
[    2.684468] RTL871X: Wrong rate section: (CCK,2T)
...
```

CGarces, 2019-11-14:
Try to reeboot?
https://forum.manjaro.org/t/solved-cant-connect-to-any-wifi-network-using-rtl8192eu-dlink-dwa-131/92023/26

I'm not sure if I should merge the 5.x.x branch, I don't now if it will cause more problems than solutions.

sargyrop, 2019-11-14:
The reboot did not help. I also tried connecting to an open WiFi to make
sure the password-check part is not the problem here. Same result, no
connection to open WiFi SSID.