Mange/rtl8192eu-linux-driver #150: AP does not run at HT speeds

yagoplx, 2019-11-19:
"iw phy" outputs that my wireless dongle's maximum rate is 300Mbps on HT, which is enabled on hostapd (Tested with 40mhz and 20mhz on WiFi n). "iw phy" also says that non-HT bitrates are 54Mbps maximum, which is the maximum speed reported by testing my phone and linux AP with iperf3.

I've also checked the output of "lsusb -t" to get a maximum rate of 480Mbps on the port.

Why does the 8192eu dongle not function at the HT bitrate that is also the one advertised in the product's packaging?
The device is a brazilian IWA 3001 by Intelbras, manufactured in China. It has a 8192EU chip and external antenna support.

Unfortunately the datasheet wouldn't be so useful here as it doesn't say much and is in Portuguese. It says however that it is MiMo and works with IEEE 802.11b/g/n modes, max 20dBm.

yagoplx, 2019-11-20:
Problem remains, but running "iwlist [interface] bitrate" shows the current bitrate as a proper 300Mb/s.

yagoplx, 2019-11-28:
I am bummed. Bought a PCI wireless card with two high gain antennas and it runs at half a megabit.

At least I have this nice wireless dongle, 50Mbps is nice.

CGarces, 2019-12-17:
Hi @yagoplx 
The patch at https://github.com/lwfinger/rtl8192ee/pull/2 apply to this driver?
You have a rtl8192eu device?

yagoplx, 2019-12-17:
Yes, I have a 8192ee and a 8192eu.

The patch was just meant to get that driver to 50Mbps max. Your driver can already reach this speed.

yagoplx, 2019-12-19:
@CGarces you might want to check this out: https://github.com/lwfinger/rtl8192ee/issues/5

It applies to your driver as well.

CGarces, 2019-12-19:
I'll check during the weekend. Feel free to send a PR if you have free time.

closed 2020-06-18 by yagoplx