Mange/rtl8192eu-linux-driver #154: Bad signal with lae

tbs3, 2019-11-27:


AlbericTrancart, 2020-02-01:
Since I updated the kernel from 4.15 to 5.3 (needed for my new GPU) I have a very weak signal too.
Don't have any clue about it though

F1xteR, 2020-12-26:
In case anyone is having issues with TL-WN8200ND V2
Right click on the Network icon.
Click Edit connections
Click on your wifi connection
Bottom left click the 3rd icon (gear)
Press the Device dropdown and change from DEVICE-NAME to DEVICE-NAME (Mac Address)
DEVICE-NAME is replaced with numbers in the dropdown, its not supposed to say DEVICE-NAME really.

mycroftsnm, 2021-01-11:
Im getting very bad speeds with TL-WN8200ND (v2) , i tried with dnsmasq (just install and enable service, no config anything) and still slow download and slow upload. @F1xteR can u guide me? im using arch and NetworkManager