Mange/rtl8192eu-linux-driver #156: Error! DKMS tree already contains: rtl8192eu-1.0

bringmemyaxe, 2019-12-11:
When I had Xubuntu 19.04, I followed instructions from [here](https://github.com/cyberkotov/mw300um-linux-driver)

SatanicRaw, 2019-12-11:
First, remove the old module, what does dkms status show?

bringmemyaxe, 2019-12-11:
```
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ dkms status
nvidia, 390.129, 5.3.0-18-generic, x86_64: installed
nvidia, 390.129, 5.3.0-24-generic, x86_64: installed
rtl8192eu, 1.0, 5.3.0-24-generic, x86_64: installed
```

What's the right command to remove the old module?

SatanicRaw, 2019-12-11:
dkms uninstall -k 5.3.0-24-generic rtl8192eu

bringmemyaxe, 2019-12-11:
```
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ sudo dkms uninstall -k 5.3.0-24-generic rtl8192eu
Error! Invalid number of arguments passed.
Usage: uninstall <module>/<module-version> or
       uninstall -m <module>/<module-version> or
       uninstall -m <module> -v <module-version>
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ sudo dkms uninstall rtl8192eu -k 5.3.0-24-generic
Error! Invalid number of arguments passed.
Usage: uninstall <module>/<module-version> or
       uninstall -m <module>/<module-version> or
       uninstall -m <module> -v <module-version>

```

SatanicRaw, 2019-12-11:
Mistaken, try as it is written in the output of DKMS
dkms uninstall -m rtl8192eu -v 1.0

bringmemyaxe, 2019-12-11:
```
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ sudo dkms uninstall rtl8192eu/1.0

-------- Uninstall Beginning --------
Module:  rtl8192eu
Version: 1.0
Kernel:  5.3.0-24-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

8192eu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.3.0-24-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.......

update-initramfs...............

DKMS: uninstall completed.
```
Then I rebooted.
And tried again:

```
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ sudo dkms add .
Error! DKMS tree already contains: rtl8192eu-1.0
You cannot add the same module/version combo more than once.
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ dkms status
nvidia, 390.129, 5.3.0-18-generic, x86_64: installed
nvidia, 390.129, 5.3.0-24-generic, x86_64: installed
rtl8192eu, 1.0, 5.3.0-24-generic, x86_64: built
robert@robert-MCP61M-M3:~/rtl8192eu-linux-driver$ sudo dkms uninstall -m rtl8192eu -v 1.0
Error! The module rtl8192eu 1.0 is not currently installed.
This module is not currently ACTIVE for kernel 5.3.0-24-generic (x86_64).
```

SatanicRaw, 2019-12-11:
Ok, try 
dkms remove rtl8192eu/1.0 --all

bringmemyaxe, 2019-12-11:
That worked and at last the driver works fine, I appreciate your help so much :)

closed 2019-12-11 by bringmemyaxe