Mange/rtl8192eu-linux-driver #157: WPA Authentication works only intermitently (Specially after reboot from Windows)

leocura, 2019-12-26:
Using this driver for the TP-WN8200ND(UN) ver. 2.0.

Sysinfo: 

> `leocura@leocura-cave:~$ cat /etc/issue
> Debian GNU/Linux bullseye/sid \n \l
> 
> leocura@leocura-cave:~$ uname -r
> 5.3.0-3-amd64
> `
> 

The device is properly recognized and it can find all networks in range. 
When I try to connect to any network, I'm given a timeout.

The password saved into wpa-supplicant is indeed correct - as the system is able to connect properly into the main network IF AND ONLY IF I reboot from Windows 10. I don't feel like the speed is impaired at all either on Linux or Windows, so I guess the only issue here is either with 8192eu or with wpa_supplicant itself. 

I've also noticed that when it DOES work, lsmod lists like this: 
(Yes, I can see the conflicting r8169 module loaded. Disabling it doesn't kill the connection, but it stops me from further connecting to any network. Raising it after disabled does not fix it either)

> `Module                  Size  Used by
> cdc_acm                40960  0
> xt_nat                 16384  220
> xt_tcpudp              20480  620
> veth                   32768  0
> xt_conntrack           16384  2
> xt_MASQUERADE          20480  202
> nf_conntrack_netlink    53248  0
> xfrm_user              45056  1
> xfrm_algo              16384  1 xfrm_user
> nft_counter            16384  653
> xt_addrtype            16384  2
> nft_compat             20480  1046
> nft_chain_nat          16384  24
> nf_nat                 49152  3 xt_nat,nft_chain_nat,xt_MASQUERADE
> nf_conntrack          167936  5 xt_conntrack,nf_nat,xt_nat,nf_conntrack_netlink,xt_MASQUERADE
> nf_defrag_ipv6         24576  1 nf_conntrack
> nf_defrag_ipv4         16384  1 nf_conntrack
> br_netfilter           28672  0
> bridge                200704  1 br_netfilter
> stp                    16384  1 bridge
> llc                    16384  2 bridge,stp
> overlay               135168  5
> nf_tables             163840  4617 nft_compat,nft_counter,nft_chain_nat
> nfnetlink              16384  4 nft_compat,nf_conntrack_netlink,nf_tables
> edac_mce_amd           32768  0
> kvm_amd               110592  0
> fuse                  139264  7
> snd_hda_codec_realtek   126976  1
> snd_hda_codec_generic    94208  1 snd_hda_codec_realtek
> ledtrig_audio          16384  2 snd_hda_codec_generic,snd_hda_codec_realtek
> snd_hda_codec_hdmi     65536  1
> snd_hda_intel          49152  4
> snd_hda_codec         159744  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
> snd_hda_core          102400  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
> snd_hwdep              16384  1 snd_hda_codec
> snd_pcm               118784  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
> serio_raw              20480  0
> pcspkr                 16384  0
> sg                     40960  0
> snd_timer              40960  1 snd_pcm
> snd                    98304  16 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm
> ccp                   102400  1 kvm_amd
> kvm                   770048  1 kvm_amd
> soundcore              16384  1 snd
> evdev                  28672  7
> sp5100_tco             20480  0
> irqbypass              16384  1 kvm
> watchdog               28672  1 sp5100_tco
> rng_core               16384  1 ccp
[... 90 lines cut ...]
> [   37.619711] R10: 0000000000000000 R11: 0000000000000001 R12: ffffbbe30134fe18
> [   37.619711] R13: ffff95a413763000 R14: ffff95a40dcd1000 R15: ffff95a40dcd1300
> [   37.619712] FS:  0000000000000000(0000) GS:ffff95a416700000(0000) knlGS:0000000000000000
> [   37.619713] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
> [   37.619714] CR2: 00007efbff440740 CR3: 00000001fe60a000 CR4: 0000000000340ee0
> [   37.619714] Call Trace:
> [   37.619749]  rtw_cfg80211_ch_switch_notify+0xe4/0x120 [8192eu]
> [   37.619770]  rtw_chk_start_clnt_join+0x64/0x70 [8192eu]
> [   37.619788]  join_cmd_hdl+0x1f3/0x340 [8192eu]
> [   37.619801]  rtw_cmd_thread+0x341/0x380 [8192eu]
> [   37.619818]  ? rtw_chk_start_clnt_join+0x70/0x70 [8192eu]
> [   37.619822]  kthread+0xf9/0x130
> [   37.619835]  ? rtw_stop_cmd_thread+0x40/0x40 [8192eu]
> [   37.619837]  ? kthread_park+0x80/0x80
> [   37.619839]  ret_from_fork+0x22/0x40
> [   37.619841] ---[ end trace f51ec2780d49e7e9 ]---
> [...]
> [ 3736.900014] libphy: r8169: probed
> [ 3736.900244] r8169 0000:04:00.0 eth0: RTL8168g/8111g, b4:2e:99:07:79:bd, XID 4c0, IRQ 37
> [ 3736.900246] r8169 0000:04:00.0 eth0: jumbo features [frames: 9200 bytes, tx checksumming: ko]
> [ 3736.902472] r8169 0000:04:00.0 enp4s0: renamed from eth0
> [ 3736.939867] r8169 0000:04:00.0: firmware: direct-loading firmware rtl_nic/rtl8168g-2.fw
> [ 3736.939951] Generic Realtek PHY r8169-400:00: attached PHY driver [Generic Realtek PHY] (mii_bus:phy_addr=r8169-400:00, irq=IGNORE)
> [ 3737.063264] r8169 0000:04:00.0 enp4s0: Link is Down
> `

yagoplx, 2019-12-26:
Is it WPA2? The Realtek drivers are commonly said to have issues with WPA1. 

The r8169 driver does not "conflict" with 8192eu, in fact, it just manages some Ethernet devices.

I have had problems with rebooting and having my wireless adapter becoming very unstable, this was solved by shutting down the computer and turning it back on manually. Later I found another way to solve it, by loading the 8192eu driver, unloading it, and loading it again. Which, well, can be easily automated by a crontab @reboot script.

leocura, 2019-12-30:
By the way, the MAC randomization fix doesn't work either.