Mange/rtl8192eu-linux-driver #158: 5.4.0-kali2-amd64 - finds networks but won't connect to any

initalwaysaye, 2020-01-12:
Hi all, hopefully someone can help me..
I just got the TL-WN823N and trying to get it working on Kali, kernel 5.4.0.
I can see wireless networks in Network Manager. if I click on one, it  asks me for key but then just times out and doesn't connect. No error messages.
i've not seen any lights flash at all on Kali on the dongle.
Works fine on same laptop on W7 partition.

so far I've tried:

- Realtek-4.4.x branch
- https://github.com/CGarces/rtl8192eu-linux-driver/tree/CGarces-patch-1
- https://github.com/Mange/rtl8192eu-linux-driver

same results whatever I try and see below further info..

**ifconfig**
wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether be:97:e5:53:7c:7a  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**lsusb**
Bus 002 Device 005: ID 0bda:0138 Realtek Semiconductor Corp. RTS5138 Card Reader Controller
Bus 002 Device 004: ID 8086:0189 Intel Corp. 
**Bus 002 Device 006: ID 2357:0109 TP-Link TL WN823N RTL8192EU**
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 04f2:b213 Chicony Electronics Co., Ltd Fujitsu Integrated Camera
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**ls /lib/modules/$(uname -r)/kernel/drivers/net/wireless**
8192eu.ko  cisco              mediatek       rsi
admtek     intel              ralink         rtl8192eu-linux-driver
ath        intersil           ray_cs.ko      wl3501_cs.ko
atmel      mac80211_hwsim.ko  realtek        zydas
broadcom   marvell            rndis_wlan.ko

**lsmod**
Module                  Size  Used by
8192eu               1212416  0

any help would be appreciated before I throw it out the window :).

thanks in advance,

mj4444ru, 2020-01-20:
Ubuntu 18.4 Kernel 5.0 is work
Ubuntu 18.4 Kernel 5.3 not work

pambrocio, 2020-01-22:
i have the same problems kali linux kernel 5.4.0

CGarces, 2020-01-22:
Hi @initalwaysaye @mj4444ru @pambrocio 
Please test #159

initalwaysaye, 2020-01-22:
Will do, thanks

initalwaysaye, 2020-01-22:
Worked for me... thanks!!

closed 2020-01-22 by initalwaysaye

mj4444ru, 2020-02-05:
This solved the problem for kernel 5.3.

PR #159 (2020-01-22): Fix for kernel 5.4
Maybe fix #158
ping @initalwaysaye for testing I don't have a rtl8192eu device.

Use git clone `https://github.com/CGarces/rtl8192eu-linux-driver/ -b fix_5.4` to test the changes.