Mange/rtl8192eu-linux-driver #161: cannot install on RPI (missing kernel headers?)

davidobrien1985, 2020-01-26:
I'm trying to install these drivers on my rpi following the instructions on this repo.

I've installed the prereqs but end up getting errors only.

```
Preparing to unpack .../raspberrypi-kernel-headers_1.20170703-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20170703-1) over (1.20170703-1) ...
Setting up raspberrypi-kernel-headers (1.20170703-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35+
run-parts: executing /etc/kernel/header_postinst.d/dkms 4.9.35-v7+
root@raspberrypi:~/rtl8192eu-linux-driver# ls /lib/modules/4.1.20-v7+/
kernel		   modules.builtin	modules.dep.bin  modules.softdep
modules.alias	   modules.builtin.bin	modules.devname  modules.symbols
modules.alias.bin  modules.dep		modules.order	 modules.symbols.bin
root@raspberrypi:~/rtl8192eu-linux-driver# dkms install rtl8192eu/1.0
Error! echo
Your kernel headers for kernel 4.1.20-v7+ cannot be found at
/lib/modules/4.1.20-v7+/build or /lib/modules/4.1.20-v7+/source.
root@raspberrypi:~/rtl8192eu-linux-driver# make install 
/bin/sh: 1: bc: not found
install -p -m 644 8192eu.ko  /lib/modules/4.1.20-v7+/kernel/drivers/net/wireless/
install: cannot stat '8192eu.ko': No such file or directory
Makefile:2304: recipe for target 'install' failed
make: *** [install] Error 1
root@raspberrypi:~/rtl8192eu-linux-driver# 
```

Not sure what I'm doing wrong.
Any tips?

Thanks!

davidobrien1985, 2020-02-09:
@Mange any idea?

Mange, 2020-02-09:
Nope. I don't use this. I'm on Rpi 3 and 4 now exclusively.

davidobrien1985, 2020-02-09:
This is on a pi3 that I'm trying to build this on.
Are you saying that on rpi3 I shouldn't need to install these drivers anymore?

Mange, 2020-02-09:
Well yeah, RPi3 has built-in wifi and Bluetooth so there is no need to buy a wifi-dongle for it or install drivers for it.

davidobrien1985, 2020-02-09:
I need a second wifi though, and every blog on this topic on the internet references this repo and following this repo's guidance doesn't work for me. 😥

Mange, 2020-02-09:
I see three errors in the output, one of which is just caused by a previous error.

1. You don't seem to have the kernel sources installed for the running version of the kernel. Maybe you need a reboot so you boot the latest kernel that you have installed?
2. The installer could not find the bc executable. Maybe you need to install that one too?

Teiby, 2020-02-10:
Used fresh installation of 2020-02-05 Raspbian Buster Lite for Raspberry Pi 1 Model B+
```
sudo apt-get update
sudo apt-get install git raspberrypi-kernel-headers build-essential dkms bc
git clone https://github.com/Mange/rtl8192eu-linux-driver
cd rtl8192eu-linux-driver/
export CONFIG_PLATFORM_I386_PC=n
export CONFIG_PLATFORM_ARM_RPI=y
sudo ln -s /usr/src/linux-headers-4.19.97+/arch/arm /usr/src/linux-headers-4.19.97+/arch/armv6l
sudo dkms add .
sudo dkms install rtl8192eu/1.0
```
It requires more steps like install `bc` and run `ln`.

CGarces, 2020-02-10:
Thanks @Teiby 
Fell free to send a PR to update  [README.md](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/README.md)
@davidobrien1985 please confirm that works on order to close the issue.

closed 2020-04-16 by davidobrien1985