Mange/rtl8192eu-linux-driver #163: Does not see the network in "N" mode.

mj4444ru, 2020-02-05:
I’m not sure what the question is at the address, but I’ll ask it.

In connection with updating the ubuntu kernel from 5.0 to 5.3, I had to install update #158, #159.

Now my rtl8192eu adapter does not see the network if it works in "Only N" mode. However, she sees her in "N/G" mode.

lshw -class network
ethernet physical wireless broadcast=yes driver=rtl8192eu multicast=yes wireless=IEEE 802.11bgn

iwlist scan
                    ESSID:"AP"
                    Protocol:IEEE 802.11bgn
                    Mode:Master
                    Frequency:2.432 GHz (Channel 5)
                    Encryption key:on
                    Bit Rates:144 Mb/s
                    Extra:rsn_ie=30140100000fac040100000fac040100000fac020c00
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    Quality=98/100  Signal level=-44 dBm  
                    Extra:fm=0003
Router Data
ath0 | 2:50:25 | 144M | 144M | HT20SGI | -41 | -95 | 54

From these data it can be seen that the router is operating in N mode.

Android and Windows normally see the network and connect to the router in "Only N" mode. Ubuntu also worked fine before updating the drivers.
What could be the problem and how to make the adapter see the network in "Only N" mode?

ZaidaZadkiel, 2020-02-07:
just to confirm it did work on my device 

just installed with 
$ ./install_wifi.sh 

dmesg ...
[  120.001647] usb 3-3: new high-speed USB device number 4 using xhci_hcd
[  120.149997] usb 3-3: New USB device found, idVendor=2c4e, idProduct=0100, bcdDevice= 2.00
[  120.149999] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  120.150001] usb 3-3: Product: 802.11n NIC 
[  120.150002] usb 3-3: Manufacturer: Realtek 
[  120.150004] usb 3-3: SerialNumber: 00e04c000001

$ uname -a
Linux zaida-medranita 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

$ iwconfig
wlx386b1c01d2f1  IEEE 802.11bgn  ESSID:"El riso 21"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.452 GHz  Access Point: 48:8D:36:F6:A4:AF   
          Bit Rate:144.4 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=98/100  Signal level=-67 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

$ readlink /sys/class/net/wlx386b1c01d2f1/device/driver
../../../../../../bus/usb/drivers/rtl8192eu

rohith7342, 2020-02-27:
hi guys the procedure which was given as per deabin based follow the instructions if the instructions dosent work go to through the commands given below
**cd rtl8192eu-linux-driver
sudo make
sudo make install 
reboot** 
after rebooting check usb of external wifi is connected or not ,check WIFI it will be connected 
the above process i tried it works successfully .after so many processes i have tried all processes for 2 days.
check it out FYIP

mj4444ru, 2020-03-30:
Ubuntu often started updating something in the bootloader and on the system.
The previous post helped me reanimate Wi-Fi.

It would be great to put this information for Debian and Ubuntu owners in "README.md"

Thanks.

CGarces, 2020-03-30:
@mj4444ru fell free to send a PR updating the documentation.

closed 2020-03-30 by CGarces

mj4444ru, 2020-04-02:
Unfortunately, I was mistaken, because I set the mode to "Only G" instead of the "Only N" mode for verification. In fact, the advice above helped make the device operational, but the device still does not see the network in "Only N" mode. Prior to the upgrade, the kernel all worked well. The system also has an old WiFi adapter, which cannot work in N mode, but still sees the router in "Only N" mode.

How can I help diagnose the problem?