Mange/rtl8192eu-linux-driver #165: kernel WARNING messages on every connect

kapamaroo, 2020-02-16:
Hi,

I am getting kernel WARNING messages on every connect on a wifi network. Apart from that, the driver seems to operate correctly afaict. Searching for warning messages from the affected functions, reveals similar issues reported online.

The USB wifi device in use, is a TP-link TL-WN822N(EU) version 5.0.

System info:
$ uname -srpv
Linux 5.3.0-28-generic #30~18.04.1-Ubuntu SMP Fri Jan 17 06:14:09 UTC 2020 x86_64

Is this a known issue with this driver and / or device?

Thanks

Copying from dmesg:

[ 1507.575852] ------------[ cut here ]------------
[ 1507.575921] WARNING: CPU: 2 PID: 32405 at /build/linux-hwe-22QmxS/linux-hwe-5.3.0/net/wireless/nl80211.c:16132 cfg80211_ch_switch_notify+0x113/0x120 [cfg80211]
[ 1507.575922] Modules linked in: 8192eu(OE) cfg80211 ccm ebtable_filter ebtables ip6table_filter ip6_tables xt_CHECKSUM iptable_mangle xt_MASQUERADE iptable_nat nf_nat nf_conntrack nf_defrag
_ipv6 nf_defrag_ipv4 libcrc32c xt_tcpudp bridge stp llc iptable_filter bpfilter nls_iso8859_1 intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irq
bypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 crypto_simd uvcvideo cryptd glue_helper snd_hda_codec_hdmi intel_cstate intel_rapl_perf snd_hda_codec_realtek s
nd_hda_codec_generic ledtrig_audio videobuf2_vmalloc snd_hda_intel videobuf2_memops snd_hda_codec snd_usb_audio videobuf2_v4l2 snd_hda_core videobuf2_common snd_usbmidi_lib joydev videodev sn
d_hwdep input_leds mc snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd mei_me mei soundcore mac_hid acpi_pad sch_fq_codel parport_pc ppdev lp binfmt_mi
sc parport ip_tables x_tables autofs4 hid_generic
[ 1507.575950]  usbhid hid i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci r8169 mxm_wmi drm libahci realtek wmi video [last unloaded: cfg80211]
[ 1507.575958] CPU: 2 PID: 32405 Comm: RTW_CMD_THREAD Tainted: G        W  OE     5.3.0-28-generic #30~18.04.1-Ubuntu
[ 1507.575959] Hardware name: MSI MS-7A74/B250M PRO-VH (MS-7A74), BIOS 1.40 04/06/2017
[ 1507.575980] RIP: 0010:cfg80211_ch_switch_notify+0x113/0x120 [cfg80211]
[ 1507.575981] Code: e4 e9 49 ff ff ff 49 8b 84 24 f0 00 00 00 48 85 c0 74 0b 48 8b 13 48 89 50 68 eb 81 0f 0b 48 c7 c7 78 b9 9b c0 e8 f0 51 d9 e0 <0f> 0b e9 6c ff ff ff 66 0f 1f 44 00 00 0f 
1f 44 00 00 55 48 89 e5
[ 1507.575982] RSP: 0000:ffffb308818bfdc0 EFLAGS: 00010282
[ 1507.575983] RAX: 0000000000000024 RBX: ffffb308818bfe00 RCX: 0000000000000000
[ 1507.575984] RDX: 0000000000000000 RSI: ffff9627eab17448 RDI: ffff9627eab17448
[ 1507.575984] RBP: ffffb308818bfde8 R08: 0000000000000476 R09: 0000000000000004
[ 1507.575985] R10: 0000000000000000 R11: 0000000000000001 R12: ffff9627e78fb800
[ 1507.575985] R13: 0000000000000001 R14: ffff9627e8444000 R15: ffff9626e7bc5000
[ 1507.575986] FS:  0000000000000000(0000) GS:ffff9627eab00000(0000) knlGS:0000000000000000
[ 1507.575987] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1507.575987] CR2: 00007fb310fa2000 CR3: 000000015eff8001 CR4: 00000000003606e0
[ 1507.575988] Call Trace:
[ 1507.576035]  rtw_cfg80211_ch_switch_notify+0xd4/0x130 [8192eu]
[ 1507.576059]  rtw_chk_start_clnt_join+0x5c/0x90 [8192eu]
[ 1507.576080]  join_cmd_hdl+0x1d7/0x340 [8192eu]
[ 1507.576095]  rtw_cmd_thread+0x34c/0x360 [8192eu]
[ 1507.576115]  ? rtw_chk_start_clnt_join+0x90/0x90 [8192eu]
[ 1507.576118]  kthread+0x121/0x140
[ 1507.576132]  ? rtw_stop_cmd_thread+0x40/0x40 [8192eu]
[ 1507.576133]  ? kthread_park+0xb0/0xb0
[ 1507.576135]  ret_from_fork+0x35/0x40
[ 1507.576137] ---[ end trace 91f5564e63426a8f ]---

[ 1507.848941] ------------[ cut here ]------------
[ 1507.849174] WARNING: CPU: 2 PID: 808 at /build/linux-hwe-22QmxS/linux-hwe-5.3.0/net/wireless/sme.c:756 __cfg80211_connect_result+0x3d3/0x450 [cfg80211]
[ 1507.849176] Modules linked in: 8192eu(OE) cfg80211 ccm ebtable_filter ebtables ip6table_filter ip6_tables xt_CHECKSUM iptable_mangle xt_MASQUERADE iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c xt_tcpudp bridge stp llc iptable_filter bpfilter nls_iso8859_1 intel_rapl_msr intel_rapl_common x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel aesni_intel aes_x86_64 crypto_simd uvcvideo cryptd glue_helper snd_hda_codec_hdmi intel_cstate intel_rapl_perf snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio videobuf2_vmalloc snd_hda_intel videobuf2_memops snd_hda_codec snd_usb_audio videobuf2_v4l2 snd_hda_core videobuf2_common snd_usbmidi_lib joydev videodev snd_hwdep input_leds mc snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq snd_seq_device snd_timer snd mei_me mei soundcore mac_hid acpi_pad sch_fq_codel parport_pc ppdev lp binfmt_misc parport ip_tables x_tables autofs4 hid_generic
[ 1507.849266]  usbhid hid i915 i2c_algo_bit drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci r8169 mxm_wmi drm libahci realtek wmi video [last unloaded: cfg80211]
[ 1507.849295] CPU: 2 PID: 808 Comm: kworker/u8:1 Tainted: G        W  OE     5.3.0-28-generic #30~18.04.1-Ubuntu
[ 1507.849297] Hardware name: MSI MS-7A74/B250M PRO-VH (MS-7A74), BIOS 1.40 04/06/2017
[ 1507.849364] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[ 1507.849426] RIP: 0010:__cfg80211_connect_result+0x3d3/0x450 [cfg80211]
[ 1507.849433] Code: c7 80 bc 9b c0 e8 a0 9a d7 e0 0f 0b 49 8b 74 24 10 48 8b 3b e8 2e 4b fd ff e9 1a fe ff ff 48 c7 c7 80 bc 9b c0 e8 80 9a d7 e0 <0f> 0b e9 07 fe ff ff 48 89 df e8 ae db fc ff e9 c1 fd ff ff e9 f5
[ 1507.849435] RSP: 0018:ffffb30881e43db0 EFLAGS: 00010282
[ 1507.849439] RAX: 0000000000000024 RBX: ffff9627e78fb800 RCX: 0000000000000000
[ 1507.849441] RDX: 0000000000000000 RSI: ffff9627eab17448 RDI: ffff9627eab17448
[ 1507.849443] RBP: ffffb30881e43e00 R08: 0000000000000492 R09: 0000000000000004
[ 1507.849445] R10: ffff9626cb08cb68 R11: 0000000000000001 R12: ffff9627dc30dc18
[ 1507.849447] R13: 0000000000000000 R14: ffffb30881e43db0 R15: dead000000000100
[ 1507.849451] FS:  0000000000000000(0000) GS:ffff9627eab00000(0000) knlGS:0000000000000000
[ 1507.849453] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1507.849456] CR2: 00007f064cd51000 CR3: 000000013f60a003 CR4: 00000000003606e0
[ 1507.849458] Call Trace:
[ 1507.849475]  ? __switch_to_asm+0x40/0x70
[ 1507.849480]  ? __switch_to_asm+0x40/0x70
[ 1507.849555]  cfg80211_process_wdev_events+0x176/0x1a0 [cfg80211]
[ 1507.849609]  ? cfg80211_process_wdev_events+0x176/0x1a0 [cfg80211]
[ 1507.849661]  cfg80211_process_rdev_events+0x38/0x80 [cfg80211]
[ 1507.849716]  cfg80211_event_work+0x1e/0x30 [cfg80211]
[ 1507.849727]  process_one_work+0x1fd/0x3f0
[ 1507.849733]  worker_thread+0x34/0x410
[ 1507.849738]  kthread+0x121/0x140
[ 1507.849743]  ? process_one_work+0x3f0/0x3f0
[ 1507.849746]  ? kthread_park+0xb0/0xb0
[ 1507.849754]  ret_from_fork+0x35/0x40
[ 1507.849761] ---[ end trace 91f5564e63426a90 ]---

CGarces, 2020-02-16:
I need to check your logs. In the meantime, you can silent the logs

Please check [this document](https://github.com/bsdfirst/rtl8814au_EW7833UAC/blob/master/Document/How_to_set_driver_debug_log_level.pdf) .
Log level can be changed on run-time so there is no need to compile a new version

danielimperial, 2020-02-26:
Same here. Module was working just fine for the TP-Link TL-WN821. For the last two days, the signal quality decreased (from 75 to 50 via `nmcli device wifi`) and a similar warning shows up in the `dmesg`:

`[fev26 13:37] ------------[ cut here ]------------
[  +0,000069] WARNING: CPU: 2 PID: 8411 at /home/kano/.cache/yay/rtl8192eu-git/src/rtl8192eu-linux-driver/hal/hal_com.c:11521 rtw_lps_state_chk+0x40/0x50 [8192eu]
[  +0,000001] Modules linked in: 8192eu(OE) cfg80211 rndis_host cdc_ether usbnet mii snd_usb_audio snd_usbmidi_lib snd_rawmidi snd_seq_device fuse gspca_ov534 gspca_main videobuf2_vmalloc v>
[  +0,000033]  ip6t_rt evdev nf_log_ipv4 nf_log_common mac_hid ipt_REJECT nf_reject_ipv4 xt_LOG xt_multiport xt_comment xt_recent xt_limit xt_addrtype xt_tcpudp xt_conntrack ip6table_filter>
[  +0,000032] CPU: 2 PID: 8411 Comm: RTW_CMD_THREAD Tainted: P        W  OE     4.19.102-1-MANJARO #1
[  +0,000001] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./Z77P-D3, BIOS F8e 11/21/2012
[  +0,000042] RIP: 0010:rtw_lps_state_chk+0x40/0x50 [8192eu]
[  +0,000003] Code: 06 00 00 48 89 ef e8 5f 9e f9 ff 84 c0 79 1d bf 01 00 00 00 e8 c1 9a fd ff 80 eb 01 75 e0 48 c7 c7 98 84 fb c1 e8 1e 21 19 cc <0f> 0b 5b 5d c3 c3 66 2e 0f 1f 84 00 00 00>
[  +0,000001] RSP: 0018:ffffa131c1117cd0 EFLAGS: 00010246
[  +0,000002] RAX: 0000000000000024 RBX: 0000000000000000 RCX: 0000000000000000
[  +0,000001] RDX: 0000000000000000 RSI: ffff8e7556b165b8 RDI: ffff8e7556b165b8
[  +0,000001] RBP: ffffa131c13e3000 R08: ffff8e7556b165b8 R09: 0000000000000001
[  +0,000001] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[  +0,000002] R13: ffffa131c1117d5c R14: ffffa131c8119000 R15: 0000000000000000
[  +0,000001] FS:  0000000000000000(0000) GS:ffff8e7556b00000(0000) knlGS:0000000000000000
[  +0,000002] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  +0,000001] CR2: 0000266c0defc000 CR3: 0000000021c0a002 CR4: 00000000000606e0
[  +0,000001] Call Trace:
[  +0,000046]  SetHwReg+0x660/0x910 [8192eu]
[  +0,000039]  SetHwReg8192E+0x53f/0x8c0 [8192eu]
[  +0,000036]  rtw_set_ps_mode+0x16e/0x210 [8192eu]
[  +0,000035]  LPS_Leave+0xa8/0xb0 [8192eu]
[  +0,000021]  traffic_status_watchdog+0x199/0x270 [8192eu]
[  +0,000032]  ? rtw_sae_preprocess+0x530/0x5c0 [8192eu]
[  +0,000020]  rtw_iface_dynamic_chk_wk_hdl+0x50/0x180 [8192eu]
[  +0,000031]  ? rtw_sae_preprocess+0x530/0x5c0 [8192eu]
[  +0,000030]  rtw_sae_preprocess+0x53a/0x5c0 [8192eu]
[  +0,000031]  rtw_sae_preprocess+0x2b8/0x5c0 [8192eu]
[  +0,000020]  rtw_dynamic_chk_wk_hdl+0xe/0x50 [8192eu]
[  +0,000021]  rtw_drvextra_cmd_hdl+0x17b/0x400 [8192eu]
[  +0,000019]  rtw_cmd_thread+0x341/0x380 [8192eu]
[  +0,000021]  ? session_tracker_cmd_hdl+0x1d0/0x1d0 [8192eu]
[  +0,000005]  kthread+0xfb/0x130
[  +0,000019]  ? rtw_stop_cmd_thread+0x40/0x40 [8192eu]
[  +0,000003]  ? kthread_park+0x80/0x80
[  +0,000003]  ret_from_fork+0x35/0x40
[  +0,000003] ---[ end trace 66a75635903bcfc9 ]---`

When this error shows up, the connection freezes and all packets are lost for the following ~20 seconds. Then, connection restores itself and the process repeats endlessly. I have absolutely no idea about what's causing it.

Output of `uname -a`:

Linux manjaro-desktop 4.19.102-1-MANJARO #1 SMP Wed Feb 5 19:48:44 UTC 2020 x86_64 GNU/Linux

urrfaust, 2020-03-13:
I have the same issue

`[ 3229.015727] ------------[ cut here ]------------
[ 3229.015854] WARNING: CPU: 1 PID: 1047 at net/wireless/nl80211.c:16135 cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
[ 3229.015855] Modules linked in: btrfs xor zstd_compress raid6_pq ufs qnx4 hfsplus hfs minix ntfs msdos jfs xfs libcrc32c nvidia_uvm(POE) nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) snd_hda_codec_realtek snd_hda_codec_generic ledtrig_audio snd_hda_codec_hdmi snd_hda_intel snd_intel_nhlt snd_hda_codec snd_usb_us122l drm_kms_helper snd_hda_core snd_usbmidi_lib drm snd_pcm snd_hwdep snd_seq_midi snd_seq_midi_event snd_rawmidi joydev ipmi_devintf input_leds ipmi_msghandler snd_seq fb_sys_fops syscopyarea snd_seq_device sysfillrect snd_timer sysimgblt snd coretemp fschmd soundcore tpm_infineon mac_hid serio_raw sch_fq_codel 8192eu(OE) cfg80211 parport_pc ppdev lp parport ip_tables x_tables autofs4 ses enclosure scsi_transport_sas hid_generic uas usbhid hid usb_storage gpio_ich psmouse i2c_i801 ahci libahci lpc_ich
[ 3229.015907] CPU: 1 PID: 1047 Comm: RTW_CMD_THREAD Tainted: P        W  OE     5.3.0-42-generic #34-Ubuntu
[ 3229.015908] Hardware name: FUJITSU SIEMENS ESPRIMO P5720                 /D2581-A1, BIOS 6.00 R1.18.2581.A1               02/05/2009
[ 3229.015939] RIP: 0010:cfg80211_ch_switch_notify.cold+0xc/0x13 [cfg80211]
[ 3229.015942] Code: ea ff ff ff e9 98 d4 fb ff 48 c7 c7 10 a5 23 c0 e8 78 9c 0d e9 0f 0b 31 c0 e9 fd e5 fb ff 48 c7 c7 10 a5 23 c0 e8 63 9c 0d e9 <0f> 0b e9 1c f6 fb ff 48 c7 c7 10 a5 23 c0 e8 50 9c 0d e9 0f 0b 66
[ 3229.015944] RSP: 0018:ffffa945c1697db0 EFLAGS: 00010246
[ 3229.015946] RAX: 0000000000000024 RBX: ffff9824160cb400 RCX: 0000000000000000
[ 3229.015948] RDX: 0000000000000000 RSI: ffff98241c117448 RDI: ffff98241c117448
[ 3229.015950] RBP: ffffa945c1697dd8 R08: ffff98241c117448 R09: 0000000000000004
[ 3229.015951] R10: 0000000000000000 R11: 0000000000000001 R12: ffffa945c1697df8
[ 3229.015953] R13: 0000000000000000 R14: ffff9824160c5000 R15: ffff98241b79d000
[ 3229.015955] FS:  0000000000000000(0000) GS:ffff98241c100000(0000) knlGS:0000000000000000
[ 3229.015957] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 3229.015959] CR2: 00007fabc4b3fff8 CR3: 0000000096b04000 CR4: 00000000000006e0
[ 3229.015961] Call Trace:
[ 3229.016047]  rtw_cfg80211_ch_switch_notify+0xe6/0x120 [8192eu]
[ 3229.016090]  rtw_chk_start_clnt_join+0x68/0x70 [8192eu]
[ 3229.016133]  join_cmd_hdl+0x218/0x360 [8192eu]
[ 3229.016163]  rtw_cmd_thread+0x33c/0x380 [8192eu]
[ 3229.016206]  ? rtw_chk_start_clnt_join+0x70/0x70 [8192eu]
[ 3229.016213]  kthread+0x104/0x140
[ 3229.016243]  ? rtw_stop_cmd_thread+0x40/0x40 [8192eu]
[ 3229.016246]  ? kthread_park+0x80/0x80
[ 3229.016251]  ret_from_fork+0x35/0x40
[ 3229.016254] ---[ end trace d7fe036321b4a2fa ]---
`