Mange/rtl8192eu-linux-driver #166: WN823N v2 not loaded as network device on Ubuntu 18.04 server

theflashpack, 2020-02-21:
I've installed the driver as per the instructions and all seemed to go well:
`$ dkms status
rtl8192eu, 1.0, 4.15.0-76-generic, x86_64: installed`

But on boot or USB insertion the device doesn't come up under `lshw -c network` or `ip a`

Prior to install it was loading the rtl8xxxu driver, now there isn't any listing.

lsusb shows the device but no device name:
```
$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 8087:0a2b Intel Corp. 
Bus 001 Device 004: ID 20a0:41e5 Clay Logic 
Bus 001 Device 007: ID 0596:053f MicroTouch Systems, Inc. 
Bus 001 Device 005: ID 1bcf:2281 Sunplus Innovation Technology Inc. 
Bus 001 Device 003: ID 2109:2812 VIA Labs, Inc. VL812 Hub
Bus 001 Device 012: ID 2357:0109 
Bus 001 Device 011: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

when I plug in the device dmesg only gives

```
[  658.333383] usb 1-2: new high-speed USB device number 12 using xhci_hcd
[  658.481967] usb 1-2: New USB device found, idVendor=2357, idProduct=0109
[  658.481971] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  658.481974] usb 1-2: Product: 802.11n NIC 
[  658.481976] usb 1-2: Manufacturer: Realtek 
[  658.481978] usb 1-2: SerialNumber: 00e04c000001
```

and udevadm gives

```
KERNEL[658.337127] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
ACTION=add
BUSNUM=001
DEVNAME=/dev/bus/usb/001/012
DEVNUM=012
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2
DEVTYPE=usb_device
MAJOR=189
MINOR=11
PRODUCT=2357/109/200
SEQNUM=5543
SUBSYSTEM=usb
TYPE=0/0/0

KERNEL[658.337774] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0 (usb)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0
DEVTYPE=usb_interface
INTERFACE=255/255/255
MODALIAS=usb:v2357p0109d0200dc00dsc00dp00icFFiscFFipFFin00
PRODUCT=2357/109/200
SEQNUM=5544
SUBSYSTEM=usb
TYPE=0/0/0

KERNEL[658.337943] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
ACTION=bind
BUSNUM=001
DEVNAME=/dev/bus/usb/001/012
DEVNUM=012
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2
DEVTYPE=usb_device
DRIVER=usb
MAJOR=189
MINOR=11
PRODUCT=2357/109/200
SEQNUM=5545
SUBSYSTEM=usb
TYPE=0/0/0

UDEV  [658.341863] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
ACTION=add
BUSNUM=001
DEVNAME=/dev/bus/usb/001/012
DEVNUM=012
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2
DEVTYPE=usb_device
DRIVER=usb
ID_BUS=usb
ID_MODEL=802.11n_NIC
ID_MODEL_ENC=802.11n\x20NIC\x20
ID_MODEL_ID=0109
ID_REVISION=0200
ID_SERIAL=Realtek_802.11n_NIC_00e04c000001
ID_SERIAL_SHORT=00e04c000001
ID_USB_INTERFACES=:ffffff:
ID_VENDOR=Realtek
ID_VENDOR_ENC=Realtek\x20
ID_VENDOR_ID=2357
MAJOR=189
MINOR=11
PRODUCT=2357/109/200
SEQNUM=5543
SUBSYSTEM=usb
TYPE=0/0/0
USEC_INITIALIZED=658341582

UDEV  [658.344910] add      /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0 (usb)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0
DEVTYPE=usb_interface
INTERFACE=255/255/255
MODALIAS=usb:v2357p0109d0200dc00dsc00dp00icFFiscFFipFFin00
PRODUCT=2357/109/200
SEQNUM=5544
SUBSYSTEM=usb
TYPE=0/0/0
USEC_INITIALIZED=658344545

UDEV  [658.346419] bind     /devices/pci0000:00/0000:00:14.0/usb1/1-2 (usb)
ACTION=bind
BUSNUM=001
DEVNAME=/dev/bus/usb/001/012
DEVNUM=012
DEVPATH=/devices/pci0000:00/0000:00:14.0/usb1/1-2
DEVTYPE=usb_device
DRIVER=usb
ID_BUS=usb
ID_MODEL=802.11n_NIC
ID_MODEL_ENC=802.11n\x20NIC\x20
ID_MODEL_ID=0109
ID_REVISION=0200
ID_SERIAL=Realtek_802.11n_NIC_00e04c000001
ID_SERIAL_SHORT=00e04c000001
ID_USB_INTERFACES=:ffffff:
ID_VENDOR=Realtek
ID_VENDOR_ENC=Realtek\x20
ID_VENDOR_ID=2357
MAJOR=189
MINOR=11
PRODUCT=2357/109/200
SEQNUM=5545
SUBSYSTEM=usb
TYPE=0/0/0
USEC_INITIALIZED=658341582
```

Is there something else I can do to check what's happening? I looked at changing the log level but there's nothing for rtl* under /proc/net

CGarces, 2020-02-21:
So I assume that the module not is loaded....
Try `sudo lshw -c network` and `lsmod`
If not is loaded, try to load it manually with modprobe. If fails loading the module should give you and error to investigate it.

theflashpack, 2020-02-21:
It seems I installed against a kernel that was replaced on reboot

At install earlier today:
```
$ sudo dkms install rtl8192eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.15.0-76-generic.........................
cleaning build area...

DKMS: build completed.

8192eu.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.15.0-76-generic/updates/dkms/

depmod...

Backing up initrd.img-4.15.0-76-generic to /boot/initrd.img-4.15.0-76-generic.old-dkms
Making new initrd.img-4.15.0-76-generic
(If next boot fails, revert to initrd.img-4.15.0-76-generic.old-dkms image)
update-initramfs.......

DKMS: install completed.
```

but I'm now on 
```
$ uname -a
Linux kiosk1 4.15.0-88-generic #88-Ubuntu SMP Tue Feb 11 20:11:34 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

I did `sudo dkms add .` but the new kernel might have already been built and waiting for reboot?

Anyway I've re-installed and rebooted and it's all working

```
*-network
       description: Wireless interface
       physical id: 2
       bus info: usb@1:2
       logical name: wlxd03745306a9c
       serial: d0:37:45:30:6a:9c
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu multicast=yes wireless=unassociated
```

thanks!

closed 2020-02-21 by theflashpack