Mange/rtl8192eu-linux-driver #167: Not installing on CentOS 7 kernel 3.10

VinayakTripathi, 2020-03-20:
Hello, I am trying to install the driver on my CentOS 7 based system but when I execute make command, it throws a ton of errors which I am not able to understand. Attaching the log file here.
Please help. Thanks.
[err.log](https://github.com/Mange/rtl8192eu-linux-driver/files/4361490/err.log)

Fachiquito, 2020-03-25:
I use Fedora and I have to adapt some parts for the system. Since CentOS and Fedora share the same lineage here what i did and hope this helps you.

**Install the DKMS**
`sudo dnf install git dkms kernel-headers @development-tools`

**Clone the repo**
`git clone https://github.com/Mange/rtl8192eu-linux-driver`

**Install the driver**
` cd rtl8192eu-linux-driver`
_then_
` sudo dkms add .` (DON'T FORGET THE DOT IN THE END)

**Compile the driver**
` sudo dkms install rtl8192eu/1.0`

**Blacklist the generic driver**
` echo "blacklist rtl8xxxu" | sudo tee /etc/modprobe.d/rtl8xxxu.conf`

**Force the drive on Boot**
` echo -e "8192eu\n\nloop" | sudo tee /etc/modules-load.d/rtl8192eu.conf`

**Update grub**
` sudo grub2-mkconfig -o /boot/grub2/grub.cfg`

**If you use EFI**
` sudo grub2-mkconfig -o /boot/efi/EFI/fedora/grub.cfg`

**Subscribe the initramfs**
` sudo dracut --force`

**Reboot the system**
` systemctl reboot -i`

balam9099, 2021-01-25:
Hi! I'm facing the same issue in Centos 8.

After execute sudo dkms install rtl8192eu/1.0 the compilation fails, here the trace error:

`/var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/os_intfs.c:1486:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, void *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, void *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  .ndo_select_queue = rtw_select_queue,
                      ^~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/os_intfs.c:1486:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:316: /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/os_intfs.o] Error 1
make[1]: *** [Makefile:1544: _module_/var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.10.1.el8_3.x86_64'
make: *** [Makefile:2293: modules] Error 2`

Any idea about the reason?

My kernel version is 4.18.

CGarces, 2021-01-25:
That's weird.
The error looks related to a change on 4.19-rc1 that should not affect 4.18. was fixed on https://github.com/Mange/rtl8192eu-linux-driver/commit/627d1adcd6cbba0303f865ff55148f45adc94775
Probably the CentOS kernel has something back-ported different than official LTS kernels.
 
Workarounds.

- Change this line from 4.19 to 4.18 https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/os_dep/linux/os_intfs.c#L1340
or...
- Use a different kernel see http://elrepo.org/tiki/tiki-index.php

If you have free time and knowledge you can send a PR with a proper patch adapting this PR

https://github.com/aircrack-ng/rtl8812au/pull/599/files

balam9099, 2021-01-25:
Hi!

I tried to change the code, and I got a new error message related to a function receiving 3 parameters but expecting 2 into an **android** named file (I'm not in no computer right know)

I also tried by upgrading the kernel using "elrepo" kernel, I got upgraded to 5.10 but then I couldn't compile the project because of "kernel-headers" for 5.10 were not available. I tried "yum update" but the library was not upgraded.

I upgraded the kernel by using "kernel-ml" I'll try with "kernel-lt" and then install "kernel-lt-devel" maybe it will work. I'll let you know as soon as I test it.

Thank you for answer me that soon.

balam9099, 2021-01-26:
Just updating.

The controller just got installed and looks like it is working fine (writing this comment using wireless connection) I'll be testing the stability this days.

In order to make it work, I had to install kernel-lt-devel and kernel-lt, then follow the instructions posted.

Thank you about your help, I missed the kernel-lt-devel installation and that is mainly why it didn't worked before.

CGarces, 2021-01-26:
@balam9099 thanks for your feedback.

closed 2021-01-26 by CGarces