Mange/rtl8192eu-linux-driver #172: Compilation errors on Raspberry Pi 4 / Fedora 32

maxnatt, 2020-04-21:
Greetings,

Thanks for your work first of all. I am using Fedora 32 (aarch64 build) on the Raspberry Pi 4 and have troubles with compiling the driver.

I have pulled latest commit of the `realtek-4.4.x` branch at the time  of writing this issue and changed `Makefile` in an appropriate way (`CONFIG_PLATFORM_ARM_AARCH64 = y` and `CONFIG_PLATFORM_I386_PC = n`).

Here's my `uname`:

`Linux box 5.6.2-301.fc32.aarch64 #1 SMP Tue Apr 7 18:10:18 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux`

Here's `make` output (same when using `dkms`):

<details>
  <summary>make</summary>

  ```sh
[root@box rtl8192eu-linux-driver]# make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.6.2-301.fc32.aarch64/build M=/home/max/rtl8192eu-linux-driver  modules
make[1]: Entering directory '/usr/src/kernels/5.6.2-301.fc32.aarch64'
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_cmd.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_security.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_debug.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_io.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_ioctl_query.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_ioctl_set.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_ieee80211.o
/home/max/rtl8192eu-linux-driver/core/rtw_ieee80211.c: In function ‘dump_ht_cap_ie’:
/home/max/rtl8192eu-linux-driver/core/rtw_ieee80211.c:1660:2: warning: ‘ht_cap_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1660 |  dump_ht_cap_ie_content(sel, ht_cap_ie + 2, ht_cap_ielen);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/max/rtl8192eu-linux-driver/core/rtw_ieee80211.c: In function ‘dump_ht_op_ie’:
/home/max/rtl8192eu-linux-driver/core/rtw_ieee80211.c:1693:2: warning: ‘ht_op_ielen’ may be used uninitialized in this function [-Wmaybe-uninitialized]
 1693 |  dump_ht_op_ie_content(sel, ht_op_ie + 2, ht_op_ielen);
      |  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_mlme.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_mlme_ext.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_mi.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_wlan_util.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_vht.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_pwrctrl.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_rf.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_chplan.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_recv.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_sta_mgt.o
  CC [M]  /home/max/rtl8192eu-linux-driver/core/rtw_ap.o
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c: In function ‘ap_free_sta’:
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3876:13: error: ‘struct sta_info’ has no member named ‘auth_len’; did you mean ‘auth_seq’?
 3876 |  if ((psta->auth_len != 0) && (psta->pauth_frame != NULL)) {
      |             ^~~~~~~~
      |             auth_seq
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3876:36: error: ‘struct sta_info’ has no member named ‘pauth_frame’
 3876 |  if ((psta->auth_len != 0) && (psta->pauth_frame != NULL)) {
      |                                    ^~
In file included from /home/max/rtl8192eu-linux-driver/include/drv_types.h:27,
                 from /home/max/rtl8192eu-linux-driver/core/rtw_ap.c:17:
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3877:17: error: ‘struct sta_info’ has no member named ‘pauth_frame’
 3877 |   rtw_mfree(psta->pauth_frame, psta->auth_len);
      |                 ^~
/home/max/rtl8192eu-linux-driver/include/osdep_service.h:261:42: note: in definition of macro ‘rtw_mfree’
  261 | #define rtw_mfree(pbuf, sz)  _rtw_mfree((pbuf), (sz))
      |                                          ^~~~
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3877:38: error: ‘struct sta_info’ has no member named ‘auth_len’; did you mean ‘auth_seq’?
 3877 |   rtw_mfree(psta->pauth_frame, psta->auth_len);
      |                                      ^~~~~~~~
/home/max/rtl8192eu-linux-driver/include/osdep_service.h:261:50: note: in definition of macro ‘rtw_mfree’
  261 | #define rtw_mfree(pbuf, sz)  _rtw_mfree((pbuf), (sz))
      |                                                  ^~
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3878:7: error: ‘struct sta_info’ has no member named ‘pauth_frame’
 3878 |   psta->pauth_frame = NULL;
      |       ^~
/home/max/rtl8192eu-linux-driver/core/rtw_ap.c:3879:9: error: ‘struct sta_info’ has no member named ‘auth_len’; did you mean ‘auth_seq’?
 3879 |   psta->auth_len = 0;
      |         ^~~~~~~~
      |         auth_seq
make[2]: *** [scripts/Makefile.build:268: /home/max/rtl8192eu-linux-driver/core/rtw_ap.o] Error 1
make[1]: *** [Makefile:1683: /home/max/rtl8192eu-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/kernels/5.6.2-301.fc32.aarch64'
make: *** [Makefile:2300: modules] Error 2
  ```
</details>

Is there any way this can be resolved?

CGarces, 2020-04-21:
I can't reproduce your error.
Please check my logs in order to find the difference between environments.
https://travis-ci.org/github/CGarces/rtl8192eu-linux-driver/jobs/677905059

I'll send a PR to add the missing `CONFIG_PLATFORM_ARM_AARCH64`
https://github.com/CGarces/rtl8192eu-linux-driver/blob/arm_64/Makefile#L134

maxnatt, 2020-04-22:
I'm curious why are there gcc errors about command line options in your build log. 🤔 

AFAIK I had the same errors only when specifying wrong architecture (e.g. when using `CONFIG_PLATFORM_ARM_RPI = y` instead of `CONFIG_PLATFORM_ARM_AARCH64`) on the Fedora aarch64, so currently build fails later when compiling files provided in the `make` log.

CGarces, 2020-04-23:
Makefile fixed at 4868187fe6fa72960bea38adb64a2ac7b92bd9f0
Ping @ertug that made the original patch

maxnatt, 2020-05-01:
Unfortunately I haven't got time to test stuff again and I was able to compile and use driver from [clnhub/rtl8192eu-linux](https://github.com/clnhub/rtl8192eu-linux) repository without any issues, so closing the issue.

CGarces, 2020-05-13:
> I'm curious why are there gcc errors about command line options in your build log. 🤔
> 
Opps, I haven't  see that...
`/bin/sh: 1: bc: not found`

Will check.

In any case, there is a problem with CONFIG_IOCTL_CFG80211.
In this line
https://github.com/Mange/rtl8192eu-linux-driver/blob/8ee79ecf880639b710a8cc7dc2dc6c39d372c857/include/sta_info.h#L482

Looks like in your compilation CONFIG_IOCTL_CFG80211 is not defined... but I don't know why.
I'll reopen the issue.

PR #181 (2020-05-20): Fix CONFIG_PLATFORM_ARM_AARCH64 compilation  
Fix CONFIG_PLATFORM_ARM_AARCH64 target
Add ARM64 to Travis build Matrix.

Fixes #179
Fixes #172
closed 2020-05-20 by CGarces