Mange/rtl8192eu-linux-driver #176: compilation error

Hugo-Trentesaux, 2020-04-29:
Compilation error:
```Makefile:1341: *** La recette commence avant la première cible. Arrêt.```
with kernel
```Kernel: 5.4.35-1-MANJARO x86_64```

CGarces, 2020-04-29:
Please provide a entire log.
Also try to force the compilation on English, maybe something like ...
`LANG=en make`

Hugo-Trentesaux, 2020-04-29:
I am using a GUI (pamac) and do not know how to do this in command line. Here is a more complete log:

```
Construction de rtl8192eu-git...
==> Création du paquet rtl8192eu-git 35.0b8394b-1 (mer. 29 avril 2020 17:53:25)
==> Vérification des dépendances pour l’exécution…
==> Vérification des dépendances pour la compilation…
==> Récupération des sources…
  -> Mise à jour du dépôt rtl8192eu-linux-driver git…
Récupération de origin
  -> fix.patch trouvé
==> Validation des fichiers source avec sha256sums…
    rtl8192eu-linux-driver ... Ignoré
    fix.patch ... Réussite
==> Suppression du répertoire $srcdir/ existant…
==> Extraction des sources…
  -> Création d’une copie de travail du dépot rtl8192eu-linux-driver git…
Clonage dans 'rtl8192eu-linux-driver'...
fait.
==> Lancement de prepare()…
patching file Makefile
Hunk #3 succeeded at 2300 (offset 957 lines).
Hunk #4 succeeded at 2304 (offset 957 lines).
/var/tmp/pamac-build-hugo/rtl8192eu-git/src
==> Lancement de pkgver()…
==> Version mise à jour : rtl8192eu-git 100.3575500-1
==> Lancement de build()…
Makefile:1341: *** La recette commence avant la première cible. Arrêt.
==> ERREUR : Une erreur s’est produite dans build().
    Abandon…
```

CGarces, 2020-04-29:
I'm not familiar with Archlinux distributions.
Are you compiling from source code (git) as described in the [readme.md](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/README.md) or are you using a archlinux AUR repository?

Hugo-Trentesaux, 2020-04-29:
I'm using this AUR: http://aur.archlinux.org/packages/rtl8192eu-git

CGarces, 2020-04-29:
Try

https://aur.archlinux.org/packages/rtl8192eu/ 
Based on the recent comments I assume that is maintained and works fine.

I case of any problem maybe you can get better support in the AUR comment section.

Hugo-Trentesaux, 2020-04-30:
The build gives me the same error. Thanks for re-orienting me. I will signal it on the AUR comments ^^

closed 2020-04-30 by CGarces