Mange/rtl8192eu-linux-driver #178: Error step 5 'sudo dkms install rtl8192eu/1.0;'

HideakiAtsuyo, 2020-05-12:
Fail at Step 5:

hideaki@Hideaki:~/Téléchargements/rtl8192eu-linux-driver-realtek-4.4.x$ sudo dkms install rtl8192eu/1.0;

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.5.0-kali2-amd64..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.5.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
```

hideaki@Hideaki:~/Téléchargements/rtl8192eu-linux-driver-realtek-4.4.x$

CGarces, 2020-05-12:
Please, post the log file.

HideakiAtsuyo, 2020-05-12:
Oh yeah sorry wait

HideakiAtsuyo, 2020-05-12:
https://pastebin.com/raw/iC5QLD9J

HideakiAtsuyo, 2020-05-12:
@CGarces this is the log file but he is available only for 1 day

CGarces, 2020-05-12:
Make sure that you are using branch `realtek-4.4.x`

HideakiAtsuyo, 2020-05-12:
Yeah i've downloaded:

https://github.com/Mange/rtl8192eu-linux-driver/tree/realtek-4.4.x

HideakiAtsuyo, 2020-05-12:
hideaki@Hideaki:~/Téléchargements/rtl8192eu-linux-driver-realtek-4.4.x$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.5.0-kali2-amd64..........(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.5.0-kali2-amd64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
hideaki@Hideaki:~/Téléchargements/rtl8192eu-linux-driver-realtek-4.4.x$ 
[Makefile(for github i put txt).txt](https://github.com/Mange/rtl8192eu-linux-driver/files/4618896/Makefile.for.github.i.put.txt.txt)
[make.log](https://github.com/Mange/rtl8192eu-linux-driver/files/4618902/make.log)

CGarces, 2020-05-13:
Hi.

I'm confused. Your makefile have this configuration
```
CONFIG_PLATFORM_ARM_AARCH64 = y
CONFIG_PLATFORM_I386_PC = n
```

Is configured to compile for arm64.
But looks like you are trying to compile for arm64... with and amd64 kernel??

`make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.5.0-kali2-amd64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
`
Are you sure that you need set `CONFIG_PLATFORM_ARM_AARCH64 = y`?

HideakiAtsuyo, 2020-05-13:
I don't know i've an amd64 kernel yes and i don't know how to config the makefile for him...

HideakiAtsuyo, 2020-05-13:
hideaki@Hideaki:~/rtl8192eu-linux-driver$ sudo dkms add .;

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.
hideaki@Hideaki:~/rtl8192eu-linux-driver$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.5.0-kali2-amd64...............................................
cleaning build area...

DKMS: build completed.

8192eu.ko:
Running module version sanity check.

Good news! Module version v5.6.4_35685.20191108_COEX20171113-0047 for 8192eu.ko
exactly matches what is already found in kernel 5.5.0-kali2-amd64.
DKMS will not replace this module.
You may override by specifying --force.

depmod...

Backing up initrd.img-5.5.0-kali2-amd64 to /boot/initrd.img-5.5.0-kali2-amd64.old-dkms
Making new initrd.img-5.5.0-kali2-amd64
(If next boot fails, revert to initrd.img-5.5.0-kali2-amd64.old-dkms image)
update-initramfs.............

DKMS: install completed.
hideaki@Hideaki:~/rtl8192eu-linux-driver$ 

When i git clone and make the stepds without modification i think that's good but i'm not good with the

```

Backing up initrd.img-5.5.0-kali2-amd64 to /boot/initrd.img-5.5.0-kali2-amd64.old-dkms
Making new initrd.img-5.5.0-kali2-amd64
(If next boot fails, revert to initrd.img-5.5.0-kali2-amd64.old-dkms image)
```
 :/

HideakiAtsuyo, 2020-05-13:
And this wont work, i use this:
wnda3100v3

lsusb(i put only the wifi key):

```
Bus 001 Device 003: ID 0846:9014 NetGear, Inc. WNDA3100v3 802.11abgn 2x2:2 [MediaTek MT7632U]
```
What is the problem?

CGarces, 2020-05-13:
So the compilation issue was caused by a wrong makefile. 

About the wifi 

`Bus 001 Device 003: ID 0846:9014 NetGear, Inc. WNDA3100v3 802.11abgn 2x2:2 [MediaTek MT7632U]`

Your device is a MT7632U. You are trying to install a wrong driver.

Try
https://github.com/jurobystricky/Netgear-A6210

closed 2020-05-13 by CGarces