Mange/rtl8192eu-linux-driver #179: Not working on Jetson TX2

marc6h, 2020-05-20:
I cannot install this driver on Jetson TX2i. Would appreciate any advice, thanks

### Configuration
In the makefile I have 'CONFIG_PLATFORM_ARM_AARCH64 = y', rest are 'n'

### Attempt 1  
$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...
Building module:
cleaning build area...
'make' all KVER=4.9.140-tegra....................(bad exit status: 2)
ERROR (dkms apport): binary package for rtl8192eu: 1.0 not found
Error! Bad return status for module build on kernel: 4.9.140-tegra (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

### Attempt 2
$ sudo make V=1 ARCH=arm64

make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.9.140-tegra/build M=/home/olyseus/rtl/rtl8192eu-linux-driver  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)

...followed by compile errors like:

error: ‘struct sta_info’ has no member named ‘auth_len’; did you mean ‘auth_seq’?
  if ((psta->auth_len != 0) && (psta->pauth_frame != NULL)) {

CGarces, 2020-05-20:
Duplicate of #172

There is a problem with CONFIG_IOCTL_CFG80211.

In this line
https://github.com/Mange/rtl8192eu-linux-driver/blob/8ee79ecf880639b710a8cc7dc2dc6c39d372c857/include/sta_info.h#L482

Looks like in your compilation **CONFIG_IOCTL_CFG80211** is not defined... but I don't know why.

¿Ideas?

CGarces, 2020-05-20:
Please try this change and give me feedback

https://github.com/CGarces/rtl8192eu-linux-driver/commit/339f8a6ed90ca733e68810934071a0e403f7f877

marc6h, 2020-05-20:
Thanks for your response. I got this:

`In file included from /home/marc/rtl2/rtl8192eu-linux-driver/include/osdep_intf.h:131:0,
                 from /home/marc/rtl2/rtl8192eu-linux-driver/include/drv_types.h:106,
                 from /home/marc/rtl2/rtl8192eu-linux-driver/os_dep/linux/ioctl_cfg80211.c:17:
/home/marc/rtl2/rtl8192eu-linux-driver/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_assoc’:
/home/marc/rtl2/rtl8192eu-linux-driver/include/../os_dep/linux/ioctl_cfg80211.h:369:60: error: implicit declaration of function ‘cfg80211_send_rx_assoc’; did you mean ‘rtw_cfg80211_send_rx_assoc’? [-Werror=implicit-function-declaration]
 #define rtw_cfg80211_send_rx_assoc(adapter, bss, buf, len) cfg80211_send_rx_assoc((adapter)->pnetdev, bss, buf, len)
                                                            ^
/home/marc/rtl2/rtl8192eu-linux-driver/os_dep/linux/ioctl_cfg80211.c:4359:3: note: in expansion of macro ‘rtw_cfg80211_send_rx_assoc’
   rtw_cfg80211_send_rx_assoc(padapter, NULL, pmgmt_frame, frame_len);
   ^~~~~~~~~~~~~~~~~~~~~~~~~~
/home/marc/rtl2/rtl8192eu-linux-driver/os_dep/linux/ioctl_cfg80211.c: In function ‘rtw_cfg80211_indicate_sta_disassoc’:
/home/marc/rtl2/rtl8192eu-linux-driver/os_dep/linux/ioctl_cfg80211.c:4418:2: error: implicit declaration of function ‘cfg80211_send_disassoc’; did you mean ‘cfg80211_abandon_assoc’? [-Werror=implicit-function-declaration]
  cfg80211_send_disassoc(padapter->pnetdev, mgmt_buf, frame_len);`

CGarces, 2020-05-20:
I'll try to reproduce the error with kernel 4.9.x tonight please be patient.

CGarces, 2020-05-20:
Please fetch last changes from #181

`make CONFIG_PLATFORM_ARM_AARCH64=y CONFIG_PLATFORM_I386_PC=n CONFIG_MP_VHT_HW_TX_MODE=n`

Looks like is working fine

You can check the build at.
https://travis-ci.org/github/Mange/rtl8192eu-linux-driver/jobs/689463337

PR #181 (2020-05-20): Fix CONFIG_PLATFORM_ARM_AARCH64 compilation  
Fix CONFIG_PLATFORM_ARM_AARCH64 target
Add ARM64 to Travis build Matrix.

Fixes #179
Fixes #172
closed 2020-05-20 by CGarces