Mange/rtl8192eu-linux-driver #187: Realtek driver version (Readme.md question)

AngryPhantom, 2020-06-30:
First of all, thank you for the driver! Works flawlessly on Debian and Void Linux.

I just have one question:

You say in your Readme.md that:

>NOTE: This branch is based on Realtek's driver versioned 4.4.1. master is based on 4.3.1.1 originally.

I see that Realtek has a newer driver on their server: ftp://files.dlink.com.au/products/DWA-131/REV_E/Drivers/DWA-131_E1_Linux_v5.6.3.1/

Is it really newer? Would you update your driver as well?

Thank you in advance.

CGarces, 2020-10-08:
Hi
Branch 4.4.x was created originally on top of 4.4.1 but updated with newer versions released by realtek.
The current version is based on [5.6.4](https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/include/rtw_version.h#L1)
Check #149 to see a changelog of the diferent versions.

closed 2020-10-08 by CGarces

AngryPhantom, 2020-10-08:
Thank you very much for your time and the info. Cheers!