Mange/rtl8192eu-linux-driver #188: Kernel fault error with Raspberry PI 3 B (kernel version 4.19.127)

ToddHwang-sw, 2020-07-15:
[WLAN] RTK Loading modules...
[   25.968575] 8192eu: loading out-of-tree module taints kernel.
[   26.312268] RTW: module init start
[   26.312746] RTW: rtl8192eu v5.6.4_35685.20191108_COEX20171113-0047
[   26.313616] RTW: build time: Jul 15 2020 13:01:24
[   26.314268] RTW: rtl8192eu BT-Coex version = COEX20171113-0047
[   26.315140] RTW: rtw_inetaddr_notifier_register
[   26.315955] RTW:
[   26.315955] usb_endpoint_descriptor(0):
[   26.316763] RTW: bLength=7
[   26.317154] RTW: bDescriptorType=5
[   26.317610] RTW: bEndpointAddress=84
[   26.318128] RTW: wMaxPacketSize=512
[   26.318605] RTW: bInterval=0
[   26.319028] RTW: RT_usb_endpoint_is_bulk_in = 4
[   26.319639] RTW:
[   26.319639] usb_endpoint_descriptor(1):
[   26.320442] RTW: bLength=7
[   26.320808] RTW: bDescriptorType=5
[   26.321297] RTW: bEndpointAddress=5
[   26.321771] RTW: wMaxPacketSize=512
[   26.322266] RTW: bInterval=0
[   26.322662] RTW: RT_usb_endpoint_is_bulk_out = 5
[   26.323311] RTW:
[   26.323311] usb_endpoint_descriptor(2):
[   26.324122] RTW: bLength=7
[   26.324493] RTW: bDescriptorType=5
[   26.324953] RTW: bEndpointAddress=6
[   26.325457] RTW: wMaxPacketSize=512
[   26.325933] RTW: bInterval=0
[   26.326350] RTW: RT_usb_endpoint_is_bulk_out = 6
[   26.326974] RTW:
[   26.326974] usb_endpoint_descriptor(3):
[   26.327771] RTW: bLength=7
[   26.328159] RTW: bDescriptorType=5
[   26.328625] RTW: bEndpointAddress=87
[   26.329138] RTW: wMaxPacketSize=64
[   26.329600] RTW: bInterval=3
[   26.330014] RTW: RT_usb_endpoint_is_int_in = 7, Interval = 3
[   26.330781] RTW:
[   26.330781] usb_endpoint_descriptor(4):
[   26.331597] RTW: bLength=7
[   26.331969] RTW: bDescriptorType=5
[   26.332457] RTW: bEndpointAddress=8
[   26.332930] RTW: wMaxPacketSize=512
[   26.333429] RTW: bInterval=0
[   26.333818] RTW: RT_usb_endpoint_is_bulk_out = 8
[   26.334468] RTW: nr_endpoint=5, in_num=2, out_num=3
[   26.334468]
[   26.335335] RTW: USB_SPEED_HIGH
[   26.335759] RTW: CHIP TYPE: RTL8192E
[   26.336622] RTW: read_chip_version_8192e 0xF0 = 0xc441135
[   26.337375] RTW: rtw_hal_config_rftype RF_Type is 2 TotalTxPath is 2
[   26.338259] RTW: Chip Version Info: CHIP_8192E_Normal_Chip_SMIC_B_CUT_2T2R_RomVer(0)
[   26.339311] RTW: _ConfigChipOutEP_8192E OutEpQueueSel(0x07), OutEpNumber(3)
[   26.340492] RTW: Boot from EFUSE, Autoload OK !
[   26.477199] RTW: HW EFUSE
[   26.477569] RTW: 0x000: 29 81 00 7C  01 40 03 00  40 74 04 50  14 00 00 00
[   26.478592] RTW: 0x010: 27 27 26 26  26 26 2A 2A  29 29 29 F2  EF EF FF FF
[   26.479597] RTW: 0x020: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.480610] RTW: 0x030: FF FF FF FF  FF FF FF FF  FF FF 2A 2A  28 28 28 28
[   26.481610] RTW: 0x040: 2B 2B 2A 29  29 F2 EF EF  FF FF FF FF  FF FF FF FF
[   26.482624] RTW: 0x050: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.483645] RTW: 0x060: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.484657] RTW: 0x070: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.485655] RTW: 0x080: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.486669] RTW: 0x090: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.487666] RTW: 0x0A0: FF FF FF FF  FF FF FF FF  FF FF FF FF  FF FF FF FF
[   26.488675] RTW: 0x0B0: FF FF FF FF  FF FF FF FF  A2 1F 20 00  00 00 FF FF
[   26.489673] RTW: 0x0C0: FF 01 00 10  00 00 00 FF  00 00 FF FF  FF FF FF FF
[... 239 lines cut ...]
[   40.783847] RTW: network.SupportedRates[2]=0B
[   40.784459] RTW: network.SupportedRates[3]=16
[   40.785058] RTW: network.SupportedRates[4]=24
[   40.785647] RTW: network.SupportedRates[5]=30
[   40.786260] RTW: network.SupportedRates[6]=48
[   40.786849] RTW: network.SupportedRates[7]=6C
[   40.787451] RTW: network.SupportedRates[8]=0C
[   40.788053] RTW: network.SupportedRates[9]=12
[   40.788638] RTW: network.SupportedRates[10]=18
[   40.789262] RTW: network.SupportedRates[11]=60
[   40.789863] RTW: bssrate_len = 12
[   40.794354] RTW: OnAssocRsp
[   40.794772] RTW: report_join_res(1, 0)
[   40.795298] RTW: rtw_joinbss_update_network
[   40.795881] RTW: +rtw_update_ht_cap()
[   40.796418] RTW: rtw_alloc_macid(wlan1) if1, mac_addr:64:e5:99:7f:3a:f2 macid:0
[   40.797411] RTW: rtw_joinbss_update_stainfo
[   40.798005] RTW: rtw_get_sta_tx_nss: 2 SS
[   40.798580] RTW: rtw_get_sta_rx_nss: 2 SS
[   40.799134] RTW: STA - MAC_ID:0, Tx - 2 SS, Rx - 2 SS
[   40.799821] RTW: STA - MAC_ID:0, SM_PS 3
[   40.800367] RTW: ht_mcs_set=ff ff 00 00, nss=2, bitmap=0000ffff
[   40.801168] RTW: ### Set STA_(0) info ###
[   40.801728] RTW: rtw_cfg80211_indicate_connect(wlan1)
[   40.802459] RTW: assoc success

yagoplx, 2020-07-16:
Please post the output of `uname -r` and `lsmod` (before connecting to a network)

-------- Mensagem Original --------

ToddHwang-sw, 2020-07-16:
Thanks for reply and your interest.

Modules ---> 

**nft_masq_ipv4 16384 1 - Live 0xffffff9aba24f000
nft_masq 16384 1 nft_masq_ipv4, Live 0xffffff9aba247000
nft_chain_nat_ipv4 16384 2 - Live 0xffffff9aba23f000
nf_nat_ipv4 20480 2 nft_masq_ipv4,nft_chain_nat_ipv4, Live 0xffffff9aba235000
nf_nat 36864 1 nf_nat_ipv4, Live 0xffffff9aba226000
nf_conntrack 135168 4 nft_masq_ipv4,nft_masq,nf_nat_ipv4,nf_nat, Live 0xffffff9aba1f6000
nf_defrag_ipv4 16384 1 nf_conntrack, Live 0xffffff9aba1ee000
nf_defrag_ipv6 20480 1 nf_conntrack, Live 0xffffff9aba1e4000
8192eu 1957888 0 - Live 0xffffff9ab9fc3000 (O)
bridge 180224 0 - Live 0xffffff9ab9f89000
brcmfmac 294912 0 - Live 0xffffff9ab9f30000
brcmutil 20480 1 brcmfmac, Live 0xffffff9ab9f26000
mac80211 528384 0 - Live 0xffffff9ab9e88000
cfg80211 372736 3 8192eu,brcmfmac,mac80211, Live 0xffffff9ab9e17000****

"Uname -a " ---> Kernel source has been downloaded from "https://github.com/raspberrypi/linux/archive/rpi-4.19.y.zip".

**Linux RPI 4.19.127 #10 SMP PREEMPT Wed Jul 15 08:10:13 PDT 2020 aarch64 GNU/Linux**

I just simply tried to activate two WiFi interfaces at the same time; broadcom built-in WiFi in raspberry pi and TP-Link USB dongle (8192eu controller) , but I am tacked down with this issue... ^^

ToddHwang-sw, 2020-07-16:
Dears. 

I checked out more deeper side of kernel and I found that the kernel register dump was come out from "WARN_ON" predicates inside of kernel code under unexpected status detection. The first one is 

15487 
15488 void cfg80211_ch_switch_notify(struct net_device *dev,
15489                                struct cfg80211_chan_def *chandef)
15490 {
15491         struct wireless_dev *wdev = dev->ieee80211_ptr;
15492         struct wiphy *wiphy = wdev->wiphy;
15493         struct cfg80211_registered_device *rdev = wiphy_to_rdev(wiphy);
15494 
15495         ASSERT_WDEV_LOCK(wdev);
15496 
15497         trace_cfg80211_ch_switch_notify(dev, chandef);
15498 
15499         wdev->chandef = *chandef;
15500         wdev->preset_chandef = *chandef;
15501 
15502         if (wdev->iftype == NL80211_IFTYPE_STATION &&
15503                                 !**WARN_ON**(!wdev->current_bss))
15504                         wdev->current_bss->pub.channel = chandef->chan;
15505 
15506         nl80211_ch_switch_notify(rdev, dev, chandef, GFP_KERNEL,
15507                                  NL80211_CMD_CH_SWITCH_NOTIFY, 0);
15508 }
15509 EXPORT_SYMBOL(cfg80211_ch_switch_notify);

Would you see why "wdev->current_bss" might have NULL value in this function. I think it should be related to WPA supplicant operation and I am now using https://w1.fi/releases/wpa_supplicant-2.9.tar.gz with a default options as follows. 

**# Driver interface for generic Linux wireless extensions
# Note: WEXT is deprecated in the current Linux kernel version and no new
# functionality is added to it. nl80211-based interface is the new
# replacement for WEXT and its use allows wpa_supplicant to properly control
# the driver to improve existing functionality like roaming and to support new
# functionality.
CONFIG_DRIVER_WEXT=y

# Driver interface for Linux drivers using the nl80211 kernel interface
CONFIG_DRIVER_NL80211=y

# Use libnl v2.0 (or 3.0) libraries.
#CONFIG_LIBNL20=y

# Use libnl 3.2 libraries (if this is selected, CONFIG_LIBNL20 is ignored)
CONFIG_LIBNL32=y

# Enable IEEE 802.1X Supplicant (automatically included if any EAP method or
# MACsec is included)
CONFIG_IEEE8021X_EAPOL=y

# EAP-MD5
CONFIG_EAP_MD5=y

# EAP-MSCHAPv2
CONFIG_EAP_MSCHAPV2=y

# EAP-TLS
CONFIG_EAP_TLS=y

# EAL-PEAP
CONFIG_EAP_PEAP=y

# EAP-TTLS
CONFIG_EAP_TTLS=y

# EAP-FAST
CONFIG_EAP_FAST=y

# EAP-GTC
CONFIG_EAP_GTC=y

# EAP-OTP
CONFIG_EAP_OTP=y

# EAP-pwd (secure authentication using only a password)
CONFIG_EAP_PWD=y

# EAP-PAX
CONFIG_EAP_PAX=y

# LEAP
CONFIG_EAP_LEAP=y

# EAP-SAKE
CONFIG_EAP_SAKE=y

# EAP-GPSK
CONFIG_EAP_GPSK=y
# Include support for optional SHA256 cipher suite in EAP-GPSK
CONFIG_EAP_GPSK_SHA256=y

# EAP-TNC and related Trusted Network Connect support (experimental)
CONFIG_EAP_TNC=y

# Wi-Fi Protected Setup (WPS)
CONFIG_WPS=y
# Enable WPS external registrar functionality
#CONFIG_WPS_ER=y
# Disable credentials for an open network by default when acting as a WPS
# registrar.
#CONFIG_WPS_REG_DISABLE_OPEN=y
# Enable WPS support with NFC config method
#CONFIG_WPS_NFC=y

# EAP-IKEv2
CONFIG_EAP_IKEV2=y

# PKCS#12 (PFX) support (used to read private key and certificate file from
# a file that usually has extension .p12 or .pfx)
CONFIG_PKCS12=y

# Smartcard support (i.e., private key on a smartcard), e.g., with openssl
# engine.
CONFIG_SMARTCARD=y

# Select control interface backend for external programs, e.g, wpa_cli:
# unix = UNIX domain sockets (default for Linux/*BSD)
# udp = UDP sockets using localhost (127.0.0.1)
# udp6 = UDP IPv6 sockets using localhost (::1)
# named_pipe = Windows Named Pipe (default for Windows)
# udp-remote = UDP sockets with remote access (only for tests systems/purpose)
# udp6-remote = UDP IPv6 sockets with remote access (only for tests purpose)
# y = use default (backwards compatibility)
# If this option is commented out, control interface is not included in the
# build.
CONFIG_CTRL_IFACE=y

# IEEE Std 802.11r-2008 (Fast BSS Transition) for station mode
CONFIG_IEEE80211R=y

# Add support for writing debug log to a file (/tmp/wpa_supplicant-log-#.txt)
CONFIG_DEBUG_FILE=y**

Somewhere in your doubt for "NULL current_bss" ??

ToddHwang-sw, 2020-07-16:
Attaching an AP from 8192eu, and while on the network, I found the following message. 
I think it should also come from driver code WARN_ON() predicates....
Raspberry PI 3 B board 4.19.127 kernel. 

 [ 2468.987069] ------------[ cut here ]------------
[ 2468.988113] WARNING: CPU: 2 PID: 223 at /home/todd/Work/linuxswtree/boards/rpi3/kernel/modules/rtl8192eu/rtl8192eu/hal/hal_com.c:11521 rtw_lps_state_chk+0x54/0x70 [8192eu]
[ 2468.990129] Modules linked in: nft_masq_ipv6 nft_chain_nat_ipv6 nf_nat_ipv6 nft_masq_ipv4 nft_masq nft_chain_nat_ipv4 nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv4 nf_defrag_ipv6 nf_tables nfnetlink 8192eu(O) bridge brcmfmac brcmutil mac80211 cfg80211
[ 2468.993115] CPU: 2 PID: 223 Comm: RTW_CMD_THREAD Tainted: G        W  O      4.19.127 #2
[ 2468.994184] Hardware name: Raspberry Pi 3 Model B Rev 1.2 (DT)
[ 2468.994960] pstate: 40000005 (nZcv daif -PAN -UAO)
[ 2468.995826] pc : rtw_lps_state_chk+0x54/0x70 [8192eu]
[ 2468.996716] lr : rtw_lps_state_chk+0x54/0x70 [8192eu]
[ 2468.997387] sp : ffffff800860ba80
[ 2468.997831] x29: ffffff800860ba80 x28: fffffff9795d0b80
[ 2468.998542] x27: 0000000000000001 x26: 0000000000000000
[ 2468.999251] x25: 0000000000000000 x24: 0000000000000000
[ 2468.999962] x23: ffffff8008e71000 x22: ffffff800860bbef
[ 2469.000672] x21: 0000000000000001 x20: ffffff8008e6b000
[ 2469.001381] x19: 0000000000000000 x18: 0000000000000010
[ 2469.002091] x17: 0000000000000000 x16: 0000000000000000
[ 2469.002802] x15: ffffffffffffffff x14: ffffff91f9288608
[ 2469.003512] x13: ffffff92792fd90f x12: ffffff91f92fd917
[ 2469.004223] x11: ffffff91f9297000 x10: ffffff800860b7b0
[ 2469.004933] x9 : 00000000ffffffd0 x8 : 2d2d2d2d2d2d2d2d
[ 2469.005643] x7 : 5d20657265682074 x6 : fffffff97abaa9e0
[ 2469.006353] x5 : fffffff97abaa9e0 x4 : 0000006781a06000
[ 2469.007062] x3 : 0000000000000001 x2 : d1f0dda3ba149d00
[ 2469.007772] x1 : 0000000000000000 x0 : 0000000000000024
[ 2469.008482] Call trace:
[ 2469.009028]  rtw_lps_state_chk+0x54/0x70 [8192eu]
[ 2469.009871]  SetHwReg+0x1ec/0x870 [8192eu]
[ 2469.010635]  SetHwReg8192E+0x54/0x810 [8192eu]
[ 2469.011445]  SetHwReg8192EU+0x28/0x130 [8192eu]
[ 2469.012263]  rtw_hal_set_hwreg+0x10/0x20 [8192eu]
[ 2469.013085]  rtw_set_ps_mode+0xa4/0x220 [8192eu]
[ 2469.013897]  LPS_Leave+0xac/0xc0 [8192eu]
[ 2469.014605]  traffic_status_watchdog+0x16c/0x260 [8192eu]
[ 2469.015494]  rtw_iface_dynamic_chk_wk_hdl+0x6c/0x1e0 [8192eu]
[ 2469.016444]  _rtw_mi_dynamic_chk_wk_hdl+0xc/0x20 [8192eu]
[ 2469.017350]  _rtw_mi_process+0x88/0xb0 [8192eu]
[ 2469.018144]  rtw_mi_dynamic_chk_wk_hdl+0x1c/0x30 [8192eu]
[ 2469.019032]  rtw_dynamic_chk_wk_hdl+0x14/0x60 [8192eu]
[ 2469.019888]  rtw_drvextra_cmd_hdl+0xc4/0x400 [8192eu]
[ 2469.020732]  rtw_cmd_thread+0x354/0x390 [8192eu]
[ 2469.021354]  kthread+0x124/0x130
[ 2469.021792]  ret_from_fork+0x10/0x1c
[ 2469.022272] ---[ end trace c309282791045984 ]---

CGarces, 2020-08-18:
Please test last version with #194 included. Reopen the issue if needed.

closed 2020-08-18 by CGarces