Mange/rtl8192eu-linux-driver #189: ad-hoc mode does not work properly

codedwrench, 2020-07-29:
I have been trying to use this driver to connect a vita up. The default driver in the linux kernel does not allow me to use ad-hoc.
This driver does seem to support it, but sadly I have been having issues with it. 

What I have been trying is just connect to it in NetworkManager. This does seem to succeed, however there is no traffic when checking in wireshark.

When I pull up wavemon I also see no channel information, leading me to believe that it didn't actually fully connect.

I know this driver is actually made by realtek itself, but I was hoping you (or someone who is reading this issue) might shed some light on this. 

[image]

yagoplx, 2020-07-29:
> [Image](https://user-images.githubusercontent.com/9693338/88780580-7c0b5c00-d18b-11ea-9940-ad7924e3c030.png)

This driver claims to support monitor mode, but, of course, that seems to have broken over the years as well. I believe it's the same situation with ad hoc.

lwfinger(+realtek) maintains the rtlwifi drivers and has stated before he doesn't care about performance. So even though we could get ad hoc support into the rtl8192ee driver in the kernel, looking at the abysmal performance of AP mode on that driver, that will be one quite terrible ad hoc experience.

This card is over a decade old and nobody seems to care about it anymore. After juggling with the 8192ee so much I just bought myself an Atheros-chipset card.

-------- Mensagem Original --------

yagoplx, 2020-07-29:
Oh, sorry, I mistook the 8192ee for the 8192eu. The situation with the kernel driver for the 8192eu is even worse. In fact, it was never tested by whom was writing the kernel drivers, and rtl8xxxu is quite lacking overall as well.

I don't know if any USB cards come with atheros chipsets, though.

-------- Mensagem Original --------