Mange/rtl8192eu-linux-driver #190: Can any RTL8192eu driver be used?

nomadmzl, 2020-08-01:
Hi, anyone knows if any rtl8192eu driver can be used? Realtek has a  [rtl8192eu driver](https://www.realtek.com/en/component/zoo/category/rtl8192eu-software) what do you think?

yagoplx, 2020-08-03:
No, it won't work. You will have to use the rtl8xxxu driver shipped with Linux or find a driver for the 8811CU elsewhere. They are not cross compatible, the 'driver' you linked to in realtek's website is actually a bundle containing various different adapter drivers for Windows.

The driver in this repository is only for USB dongles with the 8192eu chip.

-------- Mensagem Original --------

nomadmzl, 2020-08-03:
> No, it won't work. You will have to use the rtl8xxxu driver shipped with Linux or find a driver for the 8811CU elsewhere. They are not cross compatible, the 'driver' you linked to in realtek's [...]

Hi, thanks for your answer, does maybe this one from Comfast would work? http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=30&id=332

yagoplx, 2020-08-03:
That driver is unmaintained and has likely been broken for some time now. Just use the driver included in the kernel:

`sudo modprobe rtl8xxxu`

Then add it to load at boot

`echo "rtl8xxxu" | sudo tee -a /etc/modules.load.d/60-my-wifi.conf`

It's completely adequate for daily usage.

Also, you could check with `lsusb` if the computer can see if the device is plugged in the first place. It'd show up on the list.

-------- Mensagem Original --------