Mange/rtl8192eu-linux-driver #191: No Packet Injection - TxPower 12mBm

AztoreaPro, 2020-08-10:
I am having a problem getting packet injection to work and my txpower is stuck at 12mbm. 

` *root@kali:~#lshw -c network
*-network
       description: Wireless interface
       physical id: 2
       bus info: usb@2:1
       logical name: wlan0
       serial: a2:84:1b:a1:87:ec
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtl8192eu driverversion=5.7.0-kali1-amd64 multicast=yes wireless=unassociated
-------------------------------------------------------------------------------------------------------------------------------------

root@kali:~# lsusb
Bus 002 Device 002: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter

-------------------------------------------------------------------------------------------------------------------------------------

 root@kali:~# dkms status
rtl8192eu, 1.0, 5.7.0-kali1-amd64, x86_64: installed

--------------------------------------------------------------------------------------------------------------------------------------
Bus 002 Device 002: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0x818b RTL8192EU 802.11b/g/n WLAN Adapter
  bcdDevice            2.00
  iManufacturer           1 (error)
  iProduct                2 802.
  iSerial                 3 00e04c000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0035
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           5
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 802.
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x05  EP 5 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x06  EP 6 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x87  EP 7 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x000c
  bNumDeviceCaps          1
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered**

`

harlequin2, 2021-01-25:
Likewise, I have installed this driver in Kali 5.9 (USB TL-WN822N) and it installed OK, switches to monitor mode no prob, but no packet injection. aireplay-ng -9 wlan0 shows  "no Answer" to broadcast probe requests and 0/30 0% for directed probe requests. It seems the unit is not transmitting. Also, no led activity.

albaluz35, 2024-05-16:
In my case enabling monitor mode and doing packet injection WORKS! as follows:

sudo apt update
sudo apt upgrade
sudo apt-get install linux-headers-`uname -r` build-essential dkms
git clone https://github.com/Mange/rtl8192eu-linux-driver.git
cd rtl8192eu-linux
sed -i -E 's/CONFIG_WIFI_MONITOR\s*=\s*n/CONFIG_WIFI_MONITOR = y/' Makefile

sudo rmmod 8192eu rtl8xxxu r8188eu
echo "blacklist rtl8xxxu" >> ./blacklist-rtl8xxxu.conf
sudo mv ./blacklist-rtl8xxxu.conf /etc/modprobe.d/
./install_wifi.sh

-----------------------

**IF THERE ARE PROBLEMS WITH ./install_wifi.sh, RUN ./uninstall_wifi.sh then execute make commands manually:**
\# sudo make
\# sudo make install

-----------------

sudo modprobe 8192eu
sudo lsmod |grep 8192eu                                                     

!To enable Monitor mode and test packet injection:!
=================================================

WL=wlan0
sudo iwconfig
sudo ifconfig $WL down
sudo airmon-ng check kill
sudo iwconfig $WL mode monitor
sudo ifconfig $WL up
sudo iwconfig

\# Wait for a moment and retry if there are issues or reboot and try the "
!To enable Monitor mode and test packet injection:! " steps
sudo aireplay-ng --test $WL