Mange/rtl8192eu-linux-driver #197: Problem with kernel 5.8

megs-rs, 2020-08-31:
Hi there,

I had a problem with the new kernel in Fedora 32 in the last upgrade (kernel 5.8.4-200).

I see that the kernel source did a struct "sha256_state" with the same name used in this code.

I had changed to "sha256_state2".

I see too that this new kernel changed the struct "cfg80211_ops" removing the "mgmt_frame_register" member.

I don't have sure if i did the right thing, but i comment the line 9906.

And now, my drive is compiling and working good.

Best regargs,

CGarces, 2020-08-31:
**sha256_state** no longer exists,  was changed to **sha256_state_rtk** on #182
Please ensure that you fetch the last version of **realtek-4.4.x** branch.

Please note maybe #195 is needed, I haven't reviewed it.

closed 2020-08-31 by CGarces

megs-rs, 2020-09-02:
Hi CGarces,

I think the problem is the RPM package is not updated with the last source code.

I had taken the code from "https://github.com/rpmsphere/todo/raw/master/src-kmod/8192eu-4.4.1-1.src.rpm".

Thanks a lot,