Mange/rtl8192eu-linux-driver #198: On Ubuntu 18.04 stops working after kernel update.

mj4444ru, 2020-09-23:
After each update of the Ubuntu 18.04 kernel and after the subsequent reboot of the system, the WiFi adapter is no longer visible in the system.
To fix this problem, you have to run the following sequence of commands each time.
```sh
cd rtl8192eu-linux-driver-4.4.x
make clean
make
sudo make install
reboot
```

Updates have become quite frequent and this is very annoying.

What can you advise in this situation?

kelebek333, 2020-09-23:
Firstly, uninstall manually complied and installed module with following commands.

```
cd rtl8192eu-linux-driver-4.4.x
sudo make uninstall
```

After, you can try to install with DKMS. https://github.com/Mange/rtl8192eu-linux-driver#building-and-installing-using-dkms

Or you can install from PPA with following commands.

```
sudo add-apt-repository ppa:kelebek333/kablosuz
sudo apt update
sudo apt install rtl8192eu-dkms
```

mj4444ru, 2020-09-29:
The driver was originally installed via DKMS. Simply rebuilding the driver is faster than removing it from the kernel via DKMS and installing it back.

Reinstalled via PPA. If the situation does not change, I will write.

yagoplx, 2020-09-29:
The PPA is just installing the driver via dkms so it automatically builds on kernel updates.

There was no reason to remove and install manually every time with dkms, its purpose is literally to do that for you.

-------- Mensagem Original --------