Mange/rtl8192eu-linux-driver #203: GCC Error in compiling driver

hell0fri3nd, 2020-11-19:
## gcc: error: unrecognized command line option ‘-mgeneral-regs-only’

I am trying to install the driver on an Orange PI PC. As explained in the readme, I followed the steps until I got this error in the compiling phase.
I tried to do some research but noone does reference this driver, many cases included cross compiling, which is not this case I guess, since the CROSS_COMPILE variable is empty for ARM_AARCH64 platform. 

System information: `Linux orangepipc 5.8.16-sunxi #20.08.14 SMP armv7l GNU/Linux`

This is the output for `dkms install rtl8192eu/1.0;` 

```
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.8.16-sunxi...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.8.16-sunxi (armv7l)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

```

Below you can find the log file.

```
DKMS make.log for rtl8192eu-1.0 for kernel 5.8.16-sunxi (armv7l)

make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.8.16-sunxi/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.16-sunxi'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1754: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.16-sunxi'
make: *** [Makefile:2293: modules] Error 2
```
Any suggestion would be much appreciated, I am still new to the linux environment so I am struggling in finding a solution.

CGarces, 2020-11-20:
For ARM64 please try
`make CONFIG_PLATFORM_ARM_AARCH64=y CONFIG_PLATFORM_I386_PC=n CONFIG_MP_VHT_HW_TX_MODE=n`

The build has been tested over with one ubuntu arm64 image with kernel 5.3 but must work for 5.8.
https://travis-ci.org/github/Mange/rtl8192eu-linux-driver/jobs/743140972

hell0fri3nd, 2020-11-21:
I tried but it returns always the same gcc error. This is the output:
```
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.8.16-sunxi/build M=/home/hell0friend/Downloads/rtl8192eu-linux-driver  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.16-sunxi'
  CC [M]  /home/hell0friend/Downloads/rtl8192eu-linux-driver/core/rtw_cmd.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[2]: *** [scripts/Makefile.build:281: /home/hell0friend/Downloads/rtl8192eu-linux-driver/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1754: /home/hell0friend/Downloads/rtl8192eu-linux-driver] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.16-sunxi'
make: *** [Makefile:2293: modules] Error 2
```

I also checked for the `CONFIG_MP_VHT_HW_TX_MODE` parameter in the Makefile, however it was set to _n_ by default. The other two parameters were set at the beginning following the steps in the Readme file.

CGarces, 2020-11-21:
Hi

I can't reproduce the error, my only arm64 enviroment is Travis CI.

Try adding 
`EXTRA_CFLAGS += -mfloat-abi=hard`
or
`EXTRA_CFLAGS += -mfloat-abi=softfp`

To CONFIG_PLATFORM_ARM_AARCH64 at https://github.com/Mange/rtl8192eu-linux-driver/blob/realtek-4.4.x/Makefile#L1298

But honestly, i have no clue.

hell0fri3nd, 2020-11-22:
Hi, 

I tried again adding that lines of code, however it still presents the same problem. 
I reinstalled the linux headers but it did not solve this either. 

At this point I think I will substitute armbian with ubuntu and try again to compile and install the driver. 
Thank you for your support!

CGarces, 2020-11-23:
Maybe the problem is on your toolchain?

hell0fri3nd, 2020-11-27:
Hi, 
I have some (sad) news. 

I had a fresh re-install of armbian and it gave the exact error. I just tell you the passages I did and the output of the commands. When I tried to build and install the driver it raised this error: 
```
Error! Your kernel headers for kernel 5.8.16-sunxi cannot be found.
Please install the linux-headers-5.8.16-sunxi package,
```
At that point I installed the headers, but inside _/usr/src_ i found other 2 headers other than the newly installed sunxi headers. I have not the knowledge to understand if this is normal or not.
```
root@orangepipc:/usr/src# ls
linux-headers-4.15.0-124          linux-headers-5.8.16-sunxi
linux-headers-4.15.0-124-generic  rtl8192eu-1.0
```
Then, I re-tried the installation but it did not work. But this time in the log file i found a second error with the one of before:
```
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.8.16-sunxi/build M=/var/lib/dkms/rtl8192eu/1.0/bui$
make[1]: Entering directory '/usr/src/linux-headers-5.8.16-sunxi'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
gcc: error: unrecognized command line option ‘-fpatchable-function-entry=2’
scripts/Makefile.build:280: recipe for target '/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o' fa$
make[2]: *** [/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
```

I really do not know what to do. Do you think that is armbian's toolchain the problem? 

These are my make and gcc versions
```
root@orangepipc:/# make --version
GNU Make 4.1
Built for arm-unknown-linux-gnueabihf

root@orangepipc:/# gcc --version
gcc (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0

```
Thank you for your support

CGarces, 2020-12-05:
@hell0fri3nd 

You are compiling inside the orange pi? not cross compiling the code?

I don't have a arm board to reproduce the error.

hell0fri3nd, 2020-12-07:
@CGarces 
Yes, do you think I should cross compile it on my linux machine instead of compile it directly in the orange pi?

n-kall, 2020-12-07:
I'm getting this exact same issue on armbian on a NanoPi Neo
```
DKMS make.log for rtl8192eu-1.0 for kernel 5.8.16-sunxi (armv7l)
Mon Dec  7 16:12:48 EET 2020
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.8.16-sunxi/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.8.16-sunxi'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
gcc: error: unrecognized command line option '-mgeneral-regs-only'
make[2]: *** [scripts/Makefile.build:281: /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o] Error 1
make[1]: *** [Makefile:1754: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.8.16-sunxi'
make: *** [Makefile:2293: modules] Error 2
```

CGarces, 2020-12-07:
The error can be reproduced on a adm64 environment? I don't have a arm64 device.
I have compiled the current code with the LibreELEC toolchain and the Travis CI arm64 virtual machines, without reproduce the issue.

hell0fri3nd, 2021-01-06:
I actually tried to install it on Debian 5.3.5 instead of Armbian. However the problem is that there are no specific headers for this Debian version and I had to install an older one. Could I force the driver to use that specific headers version instead of the `linux-headers-$(uname -r)` version?

CGarces, 2021-01-06:
Sorry but I don't get you.

If linux-headers-$(uname -r) not exist, you have instaled a kernel not provided by debian.
On a quick look I see 4.19 for debian stable and 5.9.6 on backport

https://packages.debian.org/buster-backports/linux-headers-arm64

In any case, looks like Armbian is a better choice than plain debian.

Just give me some help to reproduce the error.

hell0fri3nd, 2021-01-06:
I used the debian stretch provided by orange pi's website. You can find it [here](https://drive.google.com/file/d/1oQtWvffEcKdJJ2833_zBvSEb2EAPbNLT/view?usp=sharing)

I am quite a newbie but from what I understood this image uses a 5.3.5+ kernel, and the only headers that I could find doing `apt-cache search linux-headers` were from linux-headers-4.9.0 to 4.19.0 

The Orange Pi PC I am using has an arm7I architecture, which is 32 bits, so I guess I cannot use debian buster with arm64 headers, right?

I followed the guide on the readme file, the only thing I changed was on the Makefile: instead of using `CONFIG_PLATFORM_ARM_AARCH64 = y`
i used
`CONFIG_PLATFORM_ARM_SUNxI = y`

Then I stopped in installing the headers, since I could not find the specific one for my version. 
I also looked for linux-headers-generic but it gave no result anyway.

CGarces, 2021-01-06:
Please consider use Armbian
Armbian already include this driver
https://github.com/armbian/build/commit/359766b1089fb8cf16c91d1db8e1df7ca45227fe
So you don't need to compile it.
https://www.armbian.com/orange-pi-pc/

hell0fri3nd, 2021-01-06:
Thank you very much! I did not know they included the driver in the last build, you really saved my day!

Thank you very much for your help 😄

closed 2021-01-06 by hell0fri3nd