Mange/rtl8192eu-linux-driver #204: Driver not working for TP-Link Archer T3U AC1300

julianYaman, 2020-11-24:
Hello, I followed the installation guide on the README on the `realtek-4.4.x` branch.
While successfully installing the driver, it does not work with my WiFi USB adapter (TP-Link Archer T3U AC1300).

This issue occurred on Arch Linux running the Kernel version `5.9.10-arch1-1`.

Currently I'm using [rtl8812au-5.6.4.2](https://github.com/gordboy/rtl8812au-5.6.4.2) with my D-Link Corp. DWA-182 Wireless AC Dualband Adapter(rev.C) WiFi adapter. (don't know if this information is necessary but I thought it might be helpful)

Is there anything I missed while installing the driver or is this even the right one to install?
I just found this repository on a Linux forum and thought this could fix my problem with my TP-Link adapter.

SatanicRaw, 2020-11-25:
Are you sure that this driver is needed for this device? Perhaps you need a driver rtl8822bu.
Show the output of the `lsusb` command

julianYaman, 2020-11-25:
Hi, here is my `lsusb` output:
```
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 002: ID 045e:02ea Microsoft Corp. Xbox One S Controller
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 05e3:0626 Genesys Logic, Inc. USB3.1 Hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 004: ID 1462:7c37 Micro Star International MYSTIC LIGHT 
Bus 003 Device 007: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 005: ID 2001:3315 D-Link Corp. DWA-182 Wireless AC Dualband Adapter(rev.C) [Realtek RTL8812AU]
Bus 003 Device 003: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 2357:0138 TP-Link 802.11ac NIC
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 05ac:0250 Apple, Inc. Aluminium Keyboard (ISO)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I'm also not sure about this being the right driver. Just saw it mentioned with TP-Link wifi adapters on some forums.

CGarces, 2020-11-25:
Hi.

lsusb is giving you the correct solution..

Bus 003 Device 005: ID 2001:3315 D-Link Corp. DWA-182 Wireless AC Dualband Adapter(rev.C) [Realtek RTL8812AU]

Please use [RTL8812AU](https://github.com/aircrack-ng/rtl8812au) 

Your device is supported
https://github.com/aircrack-ng/rtl8812au/blob/e9fbf5c051453941bbc029810b893a6c010714e6/os_dep/linux/usb_intf.c#L170

closed 2020-11-25 by CGarces

julianYaman, 2020-11-25:
Hello @CGarces, actually 
``Bus 003 Device 002: ID 2357:0138 TP-Link 802.11ac NIC`` was the device where I was looking for a driver.
My old adapter was still connected to my computer.

Nevertheless (after a long time :smile: ), I just found out that [**rtl88x2bu** (AUR)](https://aur.archlinux.org/packages/rtl88x2bu-dkms-git/) was the driver I was looking for. Looks like I was just going the wrong direction and missed this somehow - but now it is working.

Thank you very much for your responses!