Mange/rtl8192eu-linux-driver #205: Kernel 5.10 RC6 Error! Bad return status for module build on kernel: 5.10.0-051000rc6-generic (x86_64)

TheJags, 2020-12-01:
Building module fails on Kernel 5.10 RC6 in Ubuntu MATE Groovy 20.10 with the following error message:
````
admn@um:~/Downloads$ git clone https://github.com/Mange/rtl8192eu-linux-driver;

admn@um:~/Downloads$ cd rtl8192eu-linux-driver;

admn@um:~/Downloads/rtl8192eu-linux-driver$ sudo dkms add .;

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

admn@um:~/Downloads/rtl8192eu-linux-driver$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...

'make' all KVER=5.10.0-051000rc6-generic................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-051000rc6-generic (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

admn@um:~/Downloads/rtl8192eu-linux-driver$
````

/var/lib/dkms/rtl8192eu/1.0/build/make.log

````
DKMS make.log for rtl8192eu-1.0 for kernel 5.10.0-051000rc6-generic (x86_64)

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-051000rc6-generic/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules

make[1]: Entering directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
[... 88 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/rtl8192e/halrf_8192e_ce.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_regconfig8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_rtl8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e1ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e2ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.o
  MODPOST /var/lib/dkms/rtl8192eu/1.0/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.mod.o

make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko'.  Stop.

make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2

make[1]: *** [Makefile:1701: modules] Error 2

make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'

make: *** [Makefile:2293: modules] Error 2
````

Make and model of the wireless adapter: TP-Link TL WN823N USB wireless adapter.

Many thanks.

TheJags, 2020-12-01:
Ok, so I just found out that the issue I'm having could be related to this Launchpad bug:

````
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131
````
As I was having exact same issue with another drivers/modules:

https://github.com/RinCat/RTL88x2BU-Linux-Driver/issues/58

CGarces, 2020-12-01:
Hi.

It's a issue with the kernel packaging from ubuntu.
Get the missing file from git.

Quick fix;

```
wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/your_kernel_src_folder/scripts/module.lds
sudo sed -i '$ d' /usr/src/your_kernel_src_folder/scripts/module.lds
```

closed 2020-12-01 by CGarces

TheJags, 2020-12-02:
> Hi.
>
> It's a issue with the kernel packaging from ubuntu.
> Get the missing file from git.
>
> Quick fix;
>
> ```
> wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O [...]

hi @CGarces I think this may not be a case of missing file because I'm getting the same error:
````
make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko'.  Stop.
````
even after applying the steps you've detailed.

Just for the reference, there are 2 bug reports open on the issue:

Launchpad:
````
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131
````
Debian:
````
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=975571
````
The details:

1, Got the file:
````
$ sudo su
[sudo] password for admn: 
root@um:/home/admn# wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds
--2020-12-03 02:37:55--  https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.12.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.12.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 968 [text/plain]
Saving to: ‘/usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds’

/usr/src/linux-headers-5.10.0-051000rc6/scripts 100%[=====================================================================================================>]     968  --.-KB/s    in 0s      

2020-12-03 02:37:56 (33.9 MB/s) - ‘/usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds’ saved [968/968]

root@um:/home/admn# sudo sed -i '$ d' /usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds
root@um:/home/admn#
````

2, Error:
````
admn@um:~/Downloads/rtl8192eu-linux-driver$ sudo dkms add .;
[sudo] password for admn: 

Creating symlink /var/lib/dkms/rtl8192eu/1.0/source ->
                 /usr/src/rtl8192eu-1.0

DKMS: add completed.

admn@um:~/Downloads/rtl8192eu-linux-driver$ sudo dkms install rtl8192eu/1.0;

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...

'make' all KVER=5.10.0-051000rc6-generic................................................(bad exit status: 2)

Error! Bad return status for module build on kernel: 5.10.0-051000rc6-generic (x86_64)

Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
[... 128 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_mac.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_bb.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/rtl8192e/halrf_8192e_ce.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_regconfig8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_rtl8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e1ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e2ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.o
  MODPOST /var/lib/dkms/rtl8192eu/1.0/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.mod.o

make[3]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko'.  Stop.

make[2]: *** [scripts/Makefile.modpost:117: __modpost] Error 2

make[1]: *** [Makefile:1701: modules] Error 2

make[1]: Leaving directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'

make: *** [Makefile:2293: modules] Error 2
````

CGarces, 2020-12-03:
Hi.

You are downloading the file at /usr/src/linux-headers-5.10.0-051000rc6/
see : Saving to: ‘/usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds’

but the correct path is /usr/src/linux-headers-5.10.0-051000rc6-generic
See: make[1]: Entering directory '/usr/src/linux-headers-5.10.0-051000rc6-generic'

See the commit to test the change.

https://github.com/CGarces/rtl8192eu-linux-driver/commit/30370c5ae151a36b837a80573db0cb4ba63a6d24

And the build logs.

https://travis-ci.com/github/CGarces/rtl8192eu-linux-driver

TheJags, 2020-12-03:
> Hi.
>
> You are downloading the file at /usr/src/linux-headers-5.10.0-051000rc6/
> see : Saving to: ‘/usr/src/linux-headers-5.10.0-051000rc6/scripts/module.lds’
>
> but the correct path is [...]

hi @CGarces thank you so much for taking time to correct my mistake, your "quick fix" is working great :)

CGarces, 2020-12-03:
Note that https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S fetch the last version of the script.
The correct way must be download it from the correct kernel tag. Fell free to work on that and give me a hint.

TheJags, 2020-12-14:
As described on this Launchpad thread, a fix has been released by Ubuntu:

````
https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1906131
````
and the wireless module got built just fine for kernel 5.10, hence closing this issue.

Thanks alot.