Mange/rtl8192eu-linux-driver #206: error: 'struct sta_info' has no member named 'cmn'

xmapst, 2020-12-03:
```
make ARCH=arm64 CROSS_COMPILE=aarch64-openwrt-linux-musl- -C /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/linux-5.4.80 M=/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12  modules
make[5]: Entering directory '/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/linux-5.4.80'
  CC [M]  /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.o
In file included from ./include/linux/bitops.h:5,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/bitops.h:3,
                 from ./include/linux/kernel.h:12,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/kernel.h:3,
                 from ./include/linux/list.h:9,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/list.h:3,
                 from ./include/linux/module.h:9,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/basic_types.h:76,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/drv_types.h:26,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:17:
./include/linux/bits.h:8:19: error: expected identifier before '(' token
 #define BIT(nr)   (UL(1) << (nr))
                   ^
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/ieee80211.h:70:23: note: in expansion of macro 'BIT'
 #define WLAN_STA_AUTH BIT(0)
                       ^~~
/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/sta_info.h:78:2: note: in expansion of macro 'WLAN_STA_AUTH'
  WLAN_STA_AUTH,
  ^~~~~~~~~~~~~
In file included from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/asm-generic/bug.h:3,
                 from ./arch/arm64/include/asm/bug.h:26,
                 from ./include/linux/bug.h:5,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/bug.h:3,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/kernel.h:4,
                 from ./include/linux/list.h:9,
                 from /home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport/linux/list.h:3,
                 from ./include/linux/module.h:9,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/basic_types.h:76,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/drv_types.h:26,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:17:
/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/sta_info.h: In function 'sta_info_pre_move_state':
/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/sta_info.h:700:34: error: 'WLAN_STA_INSERTED' undeclared (first use in this function); did you mean 'WLAN_SP_RESERVED'?
  WARN_ON_ONCE(test_sta_flag(sta, WLAN_STA_INSERTED));
                                  ^~~~~~~~~~~~~~~~~
./include/asm-generic/bug.h:96:25: note: in definition of macro 'WARN_ON_ONCE'
  int __ret_warn_on = !!(condition);   \
                         ^~~~~~~~~
/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/sta_info.h:700:34: note: each undeclared identifier is reported only once for each function it appears in
  WARN_ON_ONCE(test_sta_flag(sta, WLAN_STA_INSERTED));
                                  ^~~~~~~~~~~~~~~~~
./include/asm-generic/bug.h:96:25: note: in definition of macro 'WARN_ON_ONCE'
  int __ret_warn_on = !!(condition);   \
                         ^~~~~~~~~
In file included from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/drv_types.h:112,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:17:
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/rtw_ap.h: At top level:
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/rtw_ap.h:45:5: warning: "CONFIG_RTW_MACADDR_ACL" is not defined, evaluates to 0 [-Wundef]
 #if CONFIG_RTW_MACADDR_ACL
     ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:17:
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/drv_types.h:1440:18: error: field 'stapriv' has incomplete type
  struct sta_priv stapriv;
                  ^~~~~~~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c: In function 'rtw_setstakey_cmd':
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:1747:40: error: 'struct sta_info' has no member named 'cmn'
  _rtw_memcpy(psetstakey_para->addr, sta->cmn.mac_addr, ETH_ALEN);
                                        ^~
In file included from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/drv_types.h:79,
                 from /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:17:
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/include/rtw_security.h:273:27: error: 'struct sta_info' has no member named 'dot118021XPrivacy'
     encry_algo = (u8) psta->dot118021XPrivacy;\
                           ^~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:1752:3: note: in expansion of macro 'GET_ENCRY_ALGO'
   GET_ENCRY_ALGO(psecuritypriv, sta, psetstakey_para->algorithm, _FALSE);
   ^~~~~~~~~~~~~~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:1758:42: error: 'struct sta_info' has no member named 'dot118021x_UncstKey'
[... 192 lines cut ...]
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5797:26: warning: initialization of 'struct sta_info *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c: In function 'rtw_setassocsta_cmdrsp_callback':
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5819:26: warning: initialization of 'struct sta_info *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
  struct sta_info *psta = rtw_get_stainfo(pstapriv, passocsta_parm->addr);
                          ^~~~~~~~~~~~~~~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5826:6: error: 'struct sta_info' has no member named 'cmn'
  psta->cmn.aid = psta->cmn.mac_id = passocsta_rsp->cam_id;
      ^~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5826:22: error: 'struct sta_info' has no member named 'cmn'
  psta->cmn.aid = psta->cmn.mac_id = passocsta_rsp->cam_id;
                      ^~
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c: In function 'session_tracker_chk_cmd':
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5204:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c: In function 'session_tracker_add_cmd':
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5209:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c: In function 'session_tracker_del_cmd':
/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12/core/rtw_cmd.c:5214:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
cc1: some warnings being treated as errors
```

CGarces, 2020-12-03:
Please try
` make CONFIG_PLATFORM_ARM_AARCH64=y CONFIG_PLATFORM_I386_PC=n`

xmapst, 2020-12-03:
> Please try
> ` make CONFIG_PLATFORM_ARM_AARCH64=y CONFIG_PLATFORM_I386_PC=n`

Thank you. I'll try

xmapst, 2020-12-04:
```
-> # grep CONFIG_PLATFORM_I386_PC package/rtl8192eu/rtl8192eu-linux-driver/Makefile
CONFIG_PLATFORM_I386_PC = n
ifeq ($(CONFIG_PLATFORM_I386_PC), y)
ifeq ($(CONFIG_PLATFORM_I386_PC), y)
-> # grep ONFIG_PLATFORM_ARM_AARCH64 package/rtl8192eu/rtl8192eu-linux-driver/Makefile
CONFIG_PLATFORM_ARM_AARCH64 = y
ifeq ($(CONFIG_PLATFORM_ARM_AARCH64), y)
```
Still fail @CGarces 
```
make -C /home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12 ARCH="arm64" CROSS_COMPILE="aarch64-openwrt-linux-musl-" KSRC="/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/linux-5.4.80" KVER="5.4.80" M="/home/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-mvebu_cortexa53/8192eu-2019-3-12" MODULE_NAME="8192eu" USER_EXTRA_CFLAGS="-DBACKPORT -DCONFIG_MINIMAL_MEMORY_USAGE  -DCONFIG_RTW_IOCTL_SET_COUNTRY -DCONFIG_CONCURRENT_MODE -DDBG_MEM_ALLOC -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -DCONFIG_PLATFORM_ARM_RPI -DCONFIG_MP_INCLUDED -DCONFIG_WEXT_PRIV -DCONFIG_WIRELESS_EXT" NOSTDINC_FLAGS="-I/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211 -I/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211/uapi -I/home/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/mac80211-backport -include backport/backport.h "
```

CGarces, 2020-12-05:
@xmapst 
Can you give me a hint about how reproduce the error. I have never used the openwrt build enviroment.

xmapst, 2020-12-06:
> @xmapst
> Can you give me a hint about how reproduce the error. I have never used the openwrt build enviroment.

There is a reproduction environment
[GL-MV1000W-Openwrt](https://github.com/xmapst/GLMV1000W-OpenWrt.git)

CGarces, 2020-12-10:
I'm not familiar with OpenWRT.
I see 3 .sh files in the repository.
How can I reproduce the error?

```
$ chmod +x offical-openwrt.sh 
$ ./offical-openwrt.sh 
```
Or there is any faster way without fetch all the dependencies?

xmapst, 2020-12-11:
> I'm not familiar with OpenWRT.
> I see 3 .sh files in the repository.
> How can I reproduce the error?
> 
> ```
> $ chmod +x offical-openwrt.sh 
> $ ./offical-openwrt.sh 
> ```
> 
> Or there is any faster way without fetch all the dependencies?

1. `sudo -E apt-get -qq updat && sudo -E apt-get -qq install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib g++-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint device-tree-compiler gperf antlr3 wget swig rsync tree` on ubuntu18.04
2. `git clone https://github.com/xmapst/GLMV1000W-OpenWrt && cd GLMV1000W-OpenWrt`
3. `git clone https://github.com/openwrt/openwrt && cd openwrt`
4. `./scripts/feeds update -a && ./scripts/feeds install -a`
5. `chmod +x ../*.sh && cp ../gl-mv1000w.config .config && make defconfig && ../offical-openwrt.sh && make -j 1 V=s`

The offical-openwrt.sh script will clone https://github.com/Mange/rtl8192eu-linux-driver to package/rtl8192eu/rtl8192eu-linux-driver, and modify CONFIG_PLATFORM_ARM_AARCH64 = y and CONFIG_PLATFORM_I386_PC = n in the Makefile

CGarces, 2020-12-11:
Thanks @xmapst 
Please continue he discussion at 
https://github.com/xmapst/GLMV1000W-OpenWrt/issues/1

CGarces, 2020-12-11:
Not related to the current source code.

closed 2020-12-11 by CGarces