Mange/rtl8192eu-linux-driver #210: I can not install this driver in 4.15.0-127 kernel

qwetqwe, 2020-12-18:
I use 4.15.0-127-generic kernel in Ubuntu 18.04.2. After I install driver by README. there is no wifi adapter found.
when I use `sudo lshw -c network` there is no wireless network.
when I use 
sudo modprobe -a 8192eu
It said : 
modprobe: ERROR: could not insert '8192eu': Unknown symbol in module, or unknown parameter
if I use dmesg it showed:
[ 2635.871889] PKCS#7 signature not signed with a trusted key
[ 2635.872332] 8192eu: Unknown symbol __cfg80211_alloc_event_skb (err 0)
[ 2635.872341] 8192eu: Unknown symbol wiphy_register (err 0)
[ 2635.872351] 8192eu: Unknown symbol cfg80211_remain_on_channel_expired (err 0)
[ 2635.872365] 8192eu: Unknown symbol cfg80211_vendor_cmd_reply (err 0)
[ 2635.872374] 8192eu: Unknown symbol cfg80211_del_sta_sinfo (err 0)
[ 2635.872383] 8192eu: Unknown symbol wiphy_unregister (err 0)
[ 2635.872402] 8192eu: Unknown symbol cfg80211_ch_switch_notify (err 0)
[ 2635.872411] 8192eu: Unknown symbol cfg80211_ch_switch_started_notify (err 0)
[ 2635.872420] 8192eu: Unknown symbol cfg80211_ready_on_channel (err 0)
[ 2635.872437] 8192eu: Unknown symbol wiphy_free (err 0)
[ 2635.872444] 8192eu: Unknown symbol wiphy_new_nm (err 0)
[ 2635.872452] 8192eu: Unknown symbol cfg80211_get_bss (err 0)
[ 2635.872462] 8192eu: Unknown symbol __cfg80211_send_event_skb (err 0)
[ 2635.872473] 8192eu: Unknown symbol cfg80211_disconnected (err 0)
[ 2635.872481] 8192eu: Unknown symbol cfg80211_michael_mic_failure (err 0)
[ 2635.872490] 8192eu: Unknown symbol cfg80211_ibss_joined (err 0)
[ 2635.872508] 8192eu: Unknown symbol cfg80211_scan_done (err 0)
[ 2635.872517] 8192eu: Unknown symbol cfg80211_roamed (err 0)
[ 2635.872524] 8192eu: Unknown symbol cfg80211_put_bss (err 0)
[ 2635.872534] 8192eu: Unknown symbol cfg80211_connect_done (err 0)
[ 2635.872547] 8192eu: Unknown symbol cfg80211_unlink_bss (err 0)
[ 2635.872559] 8192eu: Unknown symbol ieee80211_get_channel (err 0)
[ 2635.872566] 8192eu: Unknown symbol cfg80211_inform_bss_frame_data (err 0)
[ 2635.872576] 8192eu: Unknown symbol cfg80211_new_sta (err 0)
[ 2635.872590] 8192eu: Unknown symbol __cfg80211_alloc_reply_skb (err 0)
[ 2635.872598] 8192eu: Unknown symbol cfg80211_rx_mgmt (err 0)
[ 2635.872606] 8192eu: Unknown symbol ieee80211_frequency_to_channel (err 0)
[ 2635.872614] 8192eu: Unknown symbol wiphy_apply_custom_regulatory (err 0)
[ 2635.872623] 8192eu: Unknown symbol cfg80211_mgmt_tx_status (err 0)
what is the proble?

CGarces, 2020-12-18:
Looks like you are using a version compiled against a different kernel.

qwetqwe, 2020-12-20:
@CGarces 

You means that Realtek's driver is not matching with kernel version？ 
I used latest 4.4.x driver version and 4.15.0 kernel version
How can I solve it？

CGarces, 2020-12-20:
You have compiled the code against a different kernel. Compile it again.

```
sudo ./uninstall
sudo make clean
sudo ./install
```

closed 2020-12-20 by CGarces