Mange/rtl8192eu-linux-driver #212: 5.9.0-kali1-amd64 cannot be found

Gwinbleiddd, 2020-12-19:
I try install it on kali linux 2020.4 and have error after "sudo dkms install rtl8192eu/1.0"
I do all stages before this.
error:

Error! echo
Your kernel headers for kernel 5.9.0-kali1-amd64 cannot be found at
/lib/modules/5.9.0-kali1-amd64/build or /lib/modules/5.9.0-kali1-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.

CGarces, 2020-12-20:
Looks like you don't have linux-headers
try again with.

`sudo apt-get install build-essential libelf-dev linux-headers-$(uname -r)
`

Gwinbleiddd, 2020-12-20:
I did it.

Gwinbleiddd, 2020-12-20:
This don't work.
E: Unable to locate package linux-headers-5.9.0-kali1-amd64
E: Couldn't find any package by glob 'linux-headers-5.9.0-kali1-amd64'
E: Couldn't find any package by regex 'linux-headers-5.9.0-kali1-amd64'

CGarces, 2020-12-20:
So that the problem, you don't have the headers of your kernel.

closed 2020-12-20 by CGarces