Mange/rtl8192eu-linux-driver #216: Bad return status for module build on kernel: 5.10.7-arch1-1 (x86_64)

argent0, 2021-01-18:
Hi, I'm having problems compiling the module in arch. Here are the contents of `/var/lib/dkms/rtl8192eu/1.0/build/make.log` for more information.

```
DKMS make.log for rtl8192eu-1.0 for kernel 5.10.7-arch1-1 (x86_64)
Mon Jan 18 09:53:15 AM -03 2021
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.7-arch1-1/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/lib/modules/5.10.7-arch1-1/build'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘isFileReadable’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2203:11: error: implicit declaration of function ‘get_fs’; did you mean ‘get_sa’? [-Werror=implicit-function-declaration]
 2203 |   oldfs = get_fs();
      |           ^~~~~~
      |           get_sa
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2203:11: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2204:3: error: implicit declaration of function ‘set_fs’; did you mean ‘sget_fc’? [-Werror=implicit-function-declaration]
 2204 |   set_fs(KERNEL_DS);
      |   ^~~~~~
      |   sget_fc
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2204:10: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2204 |   set_fs(KERNEL_DS);
      |          ^~~~~~~~~
      |          KERNFS_NS
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2204:10: note: each undeclared identifier is reported only once for each function it appears in
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘retriveFromFile’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2241:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2241 |    oldfs = get_fs();
      |            ^~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2242:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2242 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘storeToFile’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2276:12: error: incompatible types when assigning to type ‘mm_segment_t’ from type ‘int’
 2276 |    oldfs = get_fs();
      |            ^~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:2277:11: error: ‘KERNEL_DS’ undeclared (first use in this function); did you mean ‘KERNFS_NS’?
 2277 |    set_fs(KERNEL_DS);
      |           ^~~~~~~~~
      |           KERNFS_NS
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:279: /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o] Error 1
make[1]: *** [Makefile:1805: /var/lib/dkms/rtl8192eu/1.0/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/5.10.7-arch1-1/build'
make: *** [Makefile:2293: modules] Error 2
```

This is using commit: faf68bbf82623335e7997a473f9222751e275927

CGarces, 2021-01-18:
Are you using the last source code?

The issue was solved two months ago at  https://github.com/Mange/rtl8192eu-linux-driver/commit/2ff6fb4f871819e12dd86f2177283d5c6734933c

argent0, 2021-01-18:
Thank you, that was exactly the case.

closed 2021-01-18 by argent0