Mange/rtl8192eu-linux-driver #220: Loop in authentication

guhenrique007, 2021-01-30:
When I try to connect to a wifi home network, after I enter the password, It tries to connect and open the "Authentication Required window" again, over and over...
When I try to connect through a 3G / 4G mobile internet point, it works, but home network has the reported problem.
[a print screen of problem](https://uploaddeimagens.com.br/imagens/pbOpjdw)

P.S: the same adapter and network i connected well on windows 10.

CGarces, 2021-01-30:
Maybe it's a duplicate of #57 or #46

try edit  `/etc/NetworkManager/NetworkManager.conf`
and add this lines:

```
[device]
wifi.scan-rand-mac-address=no
```

And please read the threads #57 or #46

guhenrique007, 2021-02-11:
yeah, i've already followed the solutions of these issues, but it still hasn't solved.
actually, everything that was indicated, I had already done before.

Hsirma, 2021-08-20:
I have same problem and i have tried given solutions but none of them has worked. Is there any update about this situation?