Mange/rtl8192eu-linux-driver #228: Add 8192eu to /etc/modules-load.d ??

Neilpmcd, 2021-03-28:
From the previous thread Mange/rtl8192eu-linux-driver, I followed the readme and on the newest Rasp Pi 4 Mod B (64 OS), I'm getting the following error.

If I am to add 8192eu to the /etc/modules directory, please specify the details here?

Thank-you very much,

Neil.

Makefile:681: arch/aarch64/Makefile: No such file or directory

Now, it really appears like one small step:

Building module:
cleaning build area...
'make' all KVER=5.10.17-v8+...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.17-v8+ (aarch64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

DKMS make.log for rtl8192eu-1.0 for kernel 5.10.17-v8+ (aarch64)
Sun 28 Mar 2021 02:42:27 PM CEST
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/5.10.17-v8+/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.10.17-v8+'
Makefile:681: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-5.10.17-v8+'
make: *** [Makefile:2293: modules] Error 2

CGarces, 2021-03-28:
Hi.

The make file can't find your kernel sources 

Is looking at /usr/src/linux-headers-5.10.17-v8+ but not found arch/aarch64/Makefile

Are you using a arm 32 bits kernel?

Please post the results of 

`uname -a
`
and 

`uname -r
`

Neilpmcd, 2021-03-28:
Very much appreciated:

uname -a
Linux raspberrypi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux

uname -r
5.10.17-v8+

CGarces, 2021-03-28:
Check the content of `/usr/src/linux-headers-5.10.17-v8+` looks like you don't have the correct headers

Maybe
`sudo apt-get install linux-headers-$(uname -r)`

Solve your problems

Neilpmcd, 2021-03-28:
Thanks, this is what I'm getting:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-headers-5.10.17-v8
E: Couldn't find any package by glob 'linux-headers-5.10.17-v8'
E: Couldn't find any package by regex 'linux-headers-5.10.17-v8'

CGarces, 2021-03-28:
Looks like a problem with your distribution ¿raspbian?

Please read this:

https://www.raspberrypi.org/documentation/linux/kernel/headers.md

Neilpmcd, 2021-03-28:
Thanks a lot, I'll have to get back to this tomorrow.  If you have any tips or coaching based on this, I'd greatly appreciate:

https://www.raspberrypi.org/documentation/linux/kernel/building.md

CGarces, 2021-03-28:
> Thanks a lot, I'll have to get back to this tomorrow. If you have any tips or coaching based on this, I'd greatly appreciate:
> 
> https://www.raspberrypi.org/documentation/linux/kernel/building.md

Note that you don't need to compile the entire kernel, you don't need to follow the above link

Neilpmcd, 2021-03-28:
Thanks, unsure where to begin tomorrow....in case you have ideas:

sudo apt install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20210303-1).

Neilpmcd, 2021-03-30:
Any chance there could be a fix inside the wipd gui app?  It's giving me a few choices for wpa supplicant driver?  This would be the easiest :+1: 

Otherwise, this looks much more indepth https://www.raspberrypi.org/documentation/linux/kernel/configuring.md

LinkinPony, 2022-06-14:
You can go to `/usr/src/linux-headers-5.10.17-v8+/arch/` and see if there's an folder `arm64`.
If it's there, you can modify your Makefile and add a line `ARCH = arm64`

d3vilh, 2023-02-12:
For **RaspiOS**, you need to remove kernel headers and reinstall it.

```shell
sudo apt remove raspberrypi-kernel-headers
sudo apt install dkms bc  raspberrypi-kernel-headers
```
`apt reinstall raspberrypi-kernel-headers` will not work, as it just installs headers from the repo, while when you do remove and install again - all the headers will be build from sources (what will take a lot of time), tat way you'll be sure everything is delivered.

niggowai, 2023-10-07:
got that error too on Asahi Linux.
I sneaked around by creating a symlink from aarch64 to arm64, but I think adjusting the makefile will be the smoother way

notnot, 2025-05-17:
> You can go to `/usr/src/linux-headers-5.10.17-v8+/arch/` and see if there's an folder `arm64`. If it's there, you can modify your Makefile and add a line `ARCH = arm64`

Thanks, this is what helped for me, on a totally unrelated system: Nvidia Jetson Oron Nano.