Mange/rtl8192eu-linux-driver #229: Kali 5.0 Error! echo

scoobadiver, 2021-03-28:
`sudo dkms install rtl8192eu/1.0
Error! echo
Your kernel headers for kernel 5.10.0-kali3-amd64 cannot be found at
/lib/modules/5.10.0-kali3-amd64/build or /lib/modules/5.10.0-kali3-amd64/source.
You can use the --kernelsourcedir option to tell DKMS where it's located.`

dunno whats poppin. any fix?

CGarces, 2021-03-28:
You have the kernel headers installed?

`sudo apt-get install linux-headers-$(uname -r)
`

scoobadiver, 2021-03-28:
I do

`udo apt-get install linux-headers-$(uname -r)
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
linux-headers-5.10.0-kali5-amd64 is already the newest version (5.10.24-1kali1).
linux-headers-5.10.0-kali5-amd64 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

CGarces, 2021-03-28:
I'm not familiar with kali linux I don't know where are located the kernel src files.
dkms is looking at `/lib/modules/5.10.0-kali3-amd64/build` and `/lib/modules/5.10.0-kali3-amd64/source`
You can set the correct folder manually with --kernelsourcedir
or use **make**  with the KSRC parameter

```bash
make KSRC=/path_to_kernel_headers/$(umane -r)
```

scoobadiver, 2021-03-28:
i somehow managed to make it work by rebooting the entire computer (not only the image from VM), but the adapter can only see the neighbours wi-fi (which is located right beside my bedroom), not mine (router located in the living room, maybe 5 meters from here). i really appreciatte your help on that echo error one, buddy. just wanna ask you now if you know how to improve the connection range... thx again mate :)

charliewaffles148, 2021-04-13:
Hi. Please. Help me. I will appreciate it a lot.
I had the same issue, but after rebooting, it fixed. But the problem now is:

sudo dkms install rtl8192eu/1.0;                                       

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.10.0-kali3-amd64............................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

And the log file:

DKMS make.log for rtl8192eu-1.0 for kernel 5.10.0-kali3-amd64 (x86_64)
lun 12 abr 2021 20:16:19 -05
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali3-amd64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.10.0-kali3-amd64'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/rtw_cfgvendor.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/hal_intf.o
[... 66 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/halrf_powertracking_ce.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/halrf_powertracking.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_mac.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_bb.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/rtl8192e/halrf_8192e_ce.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_regconfig8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_rtl8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e1ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e2ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.o
  MODPOST /var/lib/dkms/rtl8192eu/1.0/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.mod.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko
ld:scripts/module.lds:23: ignoring invalid character `#' in expression
[... cut ...]
make[4]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.modfinal:41: /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko] Error 1
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.modpost:117: __modpost] Error 2
make[2]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:1721: modules] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:185: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.10.0-kali3-amd64'
make: *** [Makefile:2293: modules] Error 2

CGarces, 2021-04-13:
Hi @charliewaffles148 

One file is missing on your kernel packaging
You can fetch it from linux kernel sources.

```bash
sudo wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds
sudo sed -i '$ d' /usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds

```

charliewaffles148, 2021-04-13:
Thank you for your answer. In short, the same problem. It won't happened before, I think is the new kernel update. I will appreciate any idea to solve this issue. Thanks again.

```
sudo wget https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S -O /usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds
sudo sed -i '$ d' /usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds
--2021-04-13 17:03:11--  https://raw.githubusercontent.com/torvalds/linux/master/scripts/module.lds.S
Resolviendo raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.108.133, 185.199.110.133, ...
Conectando con raw.githubusercontent.com (raw.githubusercontent.com)[185.199.109.133]:443... conectado.
Petición HTTP enviada, esperando respuesta... 200 OK
Longitud: 1132 (1,1K) [text/plain]
Grabando a: «/usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds»

/usr/src/linux-head 100%[===================>]   1,11K  --.-KB/s    en 0,03s   

2021-04-13 17:03:12 (42,7 KB/s) - «/usr/src/linux-headers-5.10.0-kali3-amd64/scripts/module.lds» guardado [1132/1132]
```

But

```
sudo dkms install rtl8192eu/1.0;                                      

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.10.0-kali3-amd64............................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-kali3-amd64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.
```

log file:

```
DKMS make.log for rtl8192eu-1.0 for kernel 5.10.0-kali3-amd64 (x86_64)
mar 13 abr 2021 17:04:12 -05
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali3-amd64/build M=/var/lib/dkms/rtl8192eu/1.0/build  modules
make[1]: se entra en el directorio '/usr/src/linux-headers-5.10.0-kali3-amd64'
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_odm.o
[... 87 lines cut ...]
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/halrf_kfree.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_mac.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_bb.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/halhwimg8192e_rf.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/halrf/rtl8192e/halrf_8192e_ce.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_regconfig8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/phydm/rtl8192e/phydm_rtl8192e.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e1ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/hal/btc/halbtc8192e2ant.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/platform/platform_ops.o
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/core/rtw_mp.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.o
  MODPOST /var/lib/dkms/rtl8192eu/1.0/build/Module.symvers
  CC [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.mod.o
  LD [M]  /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko
ld:scripts/module.lds:23: ignoring invalid character `#' in expression
ld:scripts/module.lds:29: syntax error
make[4]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.modfinal:41: /var/lib/dkms/rtl8192eu/1.0/build/8192eu.ko] Error 1
make[3]: *** [/usr/src/linux-headers-5.10.0-kali3-common/scripts/Makefile.modpost:117: __modpost] Error 2
make[2]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:1721: modules] Error 2
make[1]: *** [/usr/src/linux-headers-5.10.0-kali3-common/Makefile:185: __sub-make] Error 2
make[1]: se sale del directorio '/usr/src/linux-headers-5.10.0-kali3-amd64'
make: *** [Makefile:2293: modules] Error 2

```