Mange/rtl8192eu-linux-driver #231: Not Working On AlmaLinux 8.3 Kernel 4.18.0-240.15.1.el8_3.x86_64

jdb130496, 2021-04-01:
I have been trying to build thsi through DKMS - previously on Cent OS 8 Stream and now on Alma Linux on this kernel. I accidentlly raised a bug at AlmaLinux though this was not an OS issue - Tey helped me also by patching this one further and it was a build success (They pointed out link https://github.com/andrewlukoshko/rtl8192eu-linux-driver forked from here and patched further).  But still USB token is not recognized by OS  after plugged in USB port. USB Token DIGISOL 300Mpbs Wireless USB Adapter - DG-WN3300N. Can you help taking reference from the fork shown above? BTW, if I download Elrepo kernel it builds successfully with this repo and also USB wireless is also works. Does this one work only for Kernels above 4.4?

CGarces, 2021-04-01:
Hi

Please put the link to AlmaLinux bug and the conversation about the patching, for future reference.

Also I need the id of your usb device, please run lsusb

jdb130496, 2021-04-01:
Thank you for the response. Almalinux bug posting link: https://bugs.almalinux.org/view.php?id=47
Output of lsusb:
Bus 001 Device 004: ID 0bda:818b Realtek Semiconductor Corp. RTL8192EU 802.11b/g/n WLAN Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 046d:c077 Logitech, Inc. M105 Optical Mouse
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

jdb130496, 2021-04-01:
As you can see, the device is detected at ID: 0bda - but wi-fi in settings still shows ? sign with message no wi-fi devices attached.

jdb130496, 2021-04-01:
When build on this same OS with kernel installed from Oracle Yum Repo: Kernel - kernel-uek-5.4.17-2102.200.13.el8uek.x86_64 - DKMS build was also successful and wi-fi also started without problem - but that means I cannot use default kernel given by AlmaLinux 8.3 and also RHEL 8.3 (which is upstream of AlmaLinux 8.3).

CGarces, 2021-04-01:
Hi.

Maybe you have a problem with the drivers loaded in your system, only one can use "0bda:818b" 

With `lsmod | grep iw` maybe you can get the drivers loaded.

In theory your device must work with rtl8xxxu driver (from linux kernel) without additional modules of compiling anything.
https://github.com/torvalds/linux/blob/master/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu_core.c#L6753

If `rtl8xxxu` not works for you, maybe you need to backlist  `rtl8xxxu` to load `rtl8192eu`

Please check https://github.com/Mange/rtl8192eu-linux-driver/issues/167#issuecomment-603885572 for instructions about backlist.

Also consider use this kernels http://elrepo.org/tiki/tiki-index.php

In any case I'll add support for RHEL/CENTOS tonight (CET time).

jdb130496, 2021-04-01:
Thank you, first of all, for the detailed guidance and links. I did what you suggested above: lsmod | grep iw shows nothing. throws back to command prompt with no output. I blacklisted driver as per instructions given below repo by running - echo "blacklist rtl8xxxu" | sudo tee /etc/modprobe.d/rtl8xxxu.conf but that did not change the scenario either. I could install driver successfully through Elrepo Kernels and activated wi-fi as stated by me in the first post but since I wanted to stick to kernel provided by Centos8 Stream / RHEL 8.3 / Almalinux 8.3 by default - 4.18.240 - I was searching for solution so that I need not have to fall back on any third party kernel (including that of Oracle or Elrepo and any others) but that seems to be not working out. Thank you for trying to add support for RHEL/CENTOS as mentioned by you above. Let's hope that works out positively.

jdb130496, 2021-04-01:
I may further add that command lsmod lists all drivers which includes reference to 8192. lsmod | grep 8192eu shows as under:
8192eu               1560576  0
cfg80211              806912  1 8192eu

CGarces, 2021-04-01:
weird. So module is loaded but Wifi not works...

I'll made the patch but if your patch at https://github.com/andrewlukoshko/rtl8192eu-linux-driver not works, mine will probably also will fail.

Try `sudo dmesg` after and before plug the usb, maybe give you any hint.

jdb130496, 2021-04-01:
I attach here outputs of command sudo dmesg - before usb and after usb - here for your reference as I could not follow and get any clue from them.
[afterusb.log](https://github.com/Mange/rtl8192eu-linux-driver/files/6243899/afterusb.log)
[beforeusb.log](https://github.com/Mange/rtl8192eu-linux-driver/files/6243900/beforeusb.log)

CGarces, 2021-04-01:
Hi I see the error at rtw_os_ndevs_register but I have no idea about how fix it.

Please test my PR #232 but I think that you will get the same result.
git clone --single-branch --branch centos8_fix https://github.com/CGarces/rtl8192eu-linux-driver.git

ping @balam9099 as centos user for new ideas.

jdb130496, 2021-04-01:
Hello. Thank you for the post and error hint. I will test and post the result by tomorrow as presently I am not near that machine where I was testing this repo all the time.

jdb130496, 2021-04-01:
Meanwhile as suggested by you, I can re-look at Elrepo kernels - either ml or tl lines to look for resolution. But have they have the stability reputation as those offered in installer package by default by all these OS - AlmaLinux 8.3 or CENTOS 8.3 or RHEL 8.3?

jdb130496, 2021-04-02:
Tested the new PR#232 as suggested- Same result as earlier - 
lsmod | grep 8192eu
8192eu               1560576  0
cfg80211              806912  5 8192eu,ath9k_common,ath9k,ath,mac80211

Device built successfully with DKMS and is also detected by OS as shown in the command above but wi-fi does not start.

dmesg | grep 8192eu > rtl8192eu.log - file attached
[rtl8192eu.log](https://github.com/Mange/rtl8192eu-linux-driver/files/6247474/rtl8192eu.log)

Anyway Thank you for all your efforts in helping me. It seems I have to fall back on Elrepo kernel ml / lt - any line since both are compiling the driver successfully and wi-fi also starts. If you find it OK, please try to add 4.18.240 with red cross here for benefit of all: https://github.com/Mange/rtl8192eu-linux-driver/actions/runs/644361217 (stated above in one of your posts) 

If you still have any other idea - request to share.

May be rtl8192eu chipset - has some typical unknown problems with certain kernels versions below 4.4.

jdb130496, 2021-04-02:
Just to update - Migrated to Elrepo kernel-ml-5.11.11-1.el8.elrepo.x86_64 the latest available and DKMS builds fine and wi-fi starts working without even rebooting the system. Sharing additional information even though may not be strictly required - Only care which I need to take while I updating kernel from Elrepo is that I have to first install headers and development modules before installing Core part other wise DKMS will not be able to build the module. After that I need to install modules and then finally kernel. As for Oracle uek kernel, I first need to install development rpm (For uek kernel) and then only main uek kernel else this module will not build (There are no headers for that kernel). Before installing I need to check uname -r to confirm kernel version after rebooting system into new kernel - Just to double check. Once installed properly, DKMS takes over for the future - Any time I update the kernel - DKMS automatically builds module for that kernel provided headers and development are installed before Core.

jdb130496, 2021-04-02:
I think - time to close this thread - Thank you for all efforts and references. Regards,

jdb130496, 2021-04-02:
Just wanted to convey that if I want to use kernels of different Linux repos like Elrepo, Oracle, etc. on single OS, I have to build this module through make && make install because DKMS allows one entry at a time for each driver module - So when I tried to install for Elrepo kernel, DKMS detected previous entry of installation for Oracle kernel so it refused to install second time, but I used make&& make install so that I could use the driver on both kernels. Of course the original issue of 4.18.240 kernel failure is not resolved yet even by building it through this make&& make install compilation - but that seems to be a kernel bug.

PR #232 (2021-04-03): Add RHEL/CENTOS 8 support
Fix compilation errors with RHEL/CENTOS 8 kernels

Fix #231
closed 2021-04-02 by jdb130496