Mange/rtl8192eu-linux-driver #234: Bad return status

phpap, 2021-04-15:
After
`sudo dkms install rtl8192eu/1.0; `

Then it said 
Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=5.10.0-kali6-amd64.....(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.0-kali6-amd64 (x86_64)
Consult /var/lib/dkms/rtl8192eu/1.0/build/make.log for more information.

how to fix?

CGarces, 2021-04-15:
I'm not sure about the root cause, the error at sha256_state confuse me.

Are you using the git code from realtek-4.4.x?

sha256_state no longer exists, was changed to sha256_state_rtk on #182

Please read #197 and #199

phpap, 2021-04-16:
i didn't use it

CGarces, 2021-04-16:
> i didn't use it

You didn't use what?

phpap, 2021-04-16:
gitcode from realtek 4.4.x

phpap, 2021-04-16:
ahhh yes i use from that

CGarces, 2021-04-16:
Based on the logs provided, looks like you are using a version before #98 (2018!)
Follow the README.md and try to compile from last version (ab35c7e9672f37d75b7559758c99f6d027607687)

closed 2021-04-16 by CGarces

josegus, 2021-05-11:
I did all the steps using branch 4.4.x in Ubuntu 21.04.. is this ok? I have a lot of errors