Mange/rtl8192eu-linux-driver #237: Build fails on RHEL 8

esamson33, 2021-06-24:
Attaching log file for reference
[make.log](https://github.com/Mange/rtl8192eu-linux-driver/files/6705895/make.log)

CGarces, 2021-06-24:
I have made a patch for a similar driver.
Modify two lines and should work fine.
https://github.com/aircrack-ng/rtl8812au/compare/v5.6.4.2...CGarces:centos

esamson33, 2021-07-01:
Looking at your @CGarces patch, this one made the build on RHEL 8.4. But sorry can't test it anymore as I lost my USB TPLink
```
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index 70f6434..e623b2f 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -7484,7 +7484,7 @@ exit:
        return ret;
 }
 
-#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0)
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0)) && RHEL_RELEASE_CODE < RHEL_RELEASE_VERSION(8,0)
 static void cfg80211_rtw_mgmt_frame_register(struct wiphy *wiphy,
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 6, 0))
        struct wireless_dev *wdev,
@@ -9948,7 +9948,7 @@ static struct cfg80211_ops rtw_cfg80211_ops = {
 
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 37)) || defined(COMPAT_KERNEL_RELEASE)
        .mgmt_tx = cfg80211_rtw_mgmt_tx,
-#if LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0)
+#if (LINUX_VERSION_CODE < KERNEL_VERSION(5, 8, 0)) && RHEL_RELEASE_CODE < RHEL_RELEASE_VERSION(8,0)
        .mgmt_frame_register = cfg80211_rtw_mgmt_frame_register,
 #else
        .update_mgmt_frame_registrations = cfg80211_rtw_mgmt_frame_register,