Mange/rtl8192eu-linux-driver #240: Heavy data transfer leaks to audio device

leocura, 2021-09-05:
Not exactly a bug I believe, but whenever the device is under load I get noise over my speakers. Yeah, they have rather bad quality insulation, however the same device (tplink wm8200nd(un))does not behave that way on Windows using the stock driver. 

I'm not really sure how to log this behaviour apart from recording a vídeo of it in action. I'm running fedora 34, but the issue appears across distros. 

I'm only able to use this device(on Linux) with this driver, so going stock isn't exactly an option. While I personally don't mind this "feature"(audible network activity monitoring?! Nice), my wife is rather pissed off with my network card already.

jagmeethanspal, 2021-09-30:
That last line really raises the severity level.

leocura, 2022-02-04:
For media, yeah. But if I'm using anything that streams data from/to me I'll hear the noise loud and clear. 

I had to setup my system again, so this time I'll compile the module with a lower transmit power setting and post it up here.