Mange/rtl8192eu-linux-driver #241: Enable status LED

Celger, 2021-09-06:
Please, considerer enable the LEDs by uncommenting lines in autoconf.h as recommended  in https://github.com/Mange/rtl8192eu-linux-driver/issues/138#issuecomment-705308182 after you closed the issue. Or make an option for that.
Thanks! You are doing a great work!

CGarces, 2021-09-06:
I don't have this the device, so I can't test if the solution is valid.
Have you compiled and tested the proposed solution?

Celger, 2021-09-08:
Yeah! I have TL-WN822N(EU) ver.5.0. I uncommented the LED lines and compiled. It worked like a charm!

Uncomment the lines 152, 154 and 156 in autoconf.h

circl-lastname, 2021-10-09:
IS it necessary to uncomment the lines 154 and 156 rather than just 152?