Mange/rtl8192eu-linux-driver #242: Misbehaves on Arch

glibg10b, 2021-09-25:
After I try to turn on monitor mode with `airmon-ng start <interface>`:

When I run `sudo` (even just `sudo --help`), the terminal freezes. Neither Ctrl+C nor Ctrl+Z does anything. It also happens to network-related programs like `iw` and `NetworkManager`. At about 120 seconds after running `sudo airmon-ng start <interface>`, I start seeing errors in the journal about some of KDE Plasma's components being unresponsive. `modprobe -r` and `rmmod` freeze as well when trying to unload this module.

The same also happens with the other 8192eu modules I've tried, but not with 8xxxu. The latter unfortunately doesn't support packet injection.

Another driver I've tried (https://github.com/drygdryg/RTL8192EU) exhibits the same symptoms in Arch but works perfectly fine in Kali.

Linux version is `5.14.7-arch1-1`. Adapter is a TL-W823N v3.

glibg10b, 2021-09-25:
I switched to `5.10.68-1-lts` and now the previous problem doesn't happen anymore, but I now get:

```
Error setting channel: command failed: Device or resource busy (-16)
Error -16 likely means your card was set back to station mode by something.
Removing non-monitor wls160u4u1 interface...

You already have a wls160u4u1 device but it is NOT in station mode.
Whatever you did, don't do it again.
Please run "iw wls160u4u1 del" before attempting to continue
```

even when neither NetworkManager nor wpa_supplicant is running.

ranjith19, 2021-10-10:
Is this solved for you?

oddsito, 2022-05-17:
glibg10b you're not alone,  i've seen your questions on the arch websites asking for help and so on, i face the same issue, its because the wifi driver isn't perfectly built for Arch. We need to buy another adapter in the mean time for Arch Linux... or Use Kali yeah.

albaluz35, 2025-09-29:
this also happens to me on nethunter pro phosh/mobian when 

airmon-ng star t wlan0 $CHANNEL
sudo iw dev wlan0 set channel $CHANNEL

Error setting channel: command failed: Device or resource busy (-16)

any  clue?