Mange/rtl8192eu-linux-driver #256: Support for latest kernel versions 5.11

vijaytejak, 2021-11-12:
# Urgent and Priority
Hi,
We have been using this driver for a lot many of the system successfully thanks for that.

the kernel versions have been changing, we the driver is unable to support to the latest kernel version 5.11, had to boot the device with 5.8 version only, is there any possibility of support for 5.11, can you pls look into this

thanks in advance.

SatanicRaw, 2021-11-12:
The module supports kernels up to and including 5.15. Have you downloaded the latest version?

vijaytejak, 2021-11-12:
ohh great, will check it then, i have old version already, do i need to clone and follow the instructions as it is, or any sperate steps to follow to install the latest version on top of old version ?

SatanicRaw, 2021-11-12:
Just clone the repository and build the module, everything should work :)

CGarces, 2021-11-12:
Thanks @SatanicRaw for the support.
@vijaytejak Just follow the README.md, maybe you should use DKMS

Also you can try the rtl8xxxu module that is included in the mainline kernels, maybe support your device.

Please reopen the issue if you need more details.

closed 2021-11-12 by CGarces