Mange/rtl8192eu-linux-driver #258: Thanks @SatanicRaw for the support.

vijaytejak, 2021-11-24:
Thanks @SatanicRaw for the support.
@vijaytejak Just follow the README.md, maybe you should use DKMS

Also you can try the rtl8xxxu module that is included in the mainline kernels, maybe support your device.

Please reopen the issue if you need more details.

_Originally posted by @CGarces in https://github.com/Mange/rtl8192eu-linux-driver/issues/256#issuecomment-967038786_

vijaytejak, 2021-11-24:
Hi, can you pls guide, what specific to consider for updating the driver, just close and install the same steps or anything additional ?

SatanicRaw, 2021-11-26:
Hi. If you used the command `git clone  https://github.com/Mange/rtl8192eu-linux-driver`
then just run the `git pull` command in the **rtl8192eu-linux-driver** directory, it will update the module to the latest version in the repository.
Just before building the updated module, do not forget to run the `make clean` command.

CGarces, 2021-11-26:
Please read de Readme.md file
"Building and installing using DKMS"

vijaytejak, 2021-11-26:
unfortunately, the directories were deleted, let me know how to update in this case.

thank you

SatanicRaw, 2021-11-26:
`git clone https://github.com/Mange/rtl8192eu-linux-driver`

vijaytejak, 2021-11-26:
ok,so git-clone and fresh install will do the work?

SatanicRaw, 2021-11-26:
Yes.

closed 2022-09-10 by CGarces