Mange/rtl8192eu-linux-driver #259: Performance issues

vijaytejak, 2021-11-24:
Hi all,

I am using D-Link DWA 131 WiFi adapter for Ubuntu 20.4 OS.
I use Unifi NanoHD Access Points, i have 150 ubuntu devices with this specs separated into multiple rooms with NanoHD ,AC PRO APs
the clients wont proper internet it will be slow, disconnections, getting question mark, slow loading very frequently we are suffering with this configurations a lot, the other laptops/ipads are doing great, because they are not using D-Link adapter, if the same adapter is used with Windows they are working great, the channel utilization of the APs will always be high and causing slow internet speeds

I feel something has to be done in the driver it self, for getting better performance

# Please Help me

vijaytejak, 2021-11-24:
@CGarces we can together work on this to resolve the issue, of course i cant code, but can help in all other aspects

CGarces, 2021-11-24:
Hi.

First of all please execute lsusb and give me the usb id of your wifi device.

1. Try to use mainline driver, rtl8xxxu is your best option.
2. If option 1 not works try to identify the problem, maybe a patch can be send to mainline kernel.
3. If you must use this driver, try to build if first from commit fa64647e38c1bed895ea74f08fb44dabc797f1a2 and confirm the issue, maybe I broke something in the last builds.

vijaytejak, 2021-11-25:
Here is the lsusb command output of WiFi driver, let me know command to check, if this is wrong
```Bus 001 Device 002: ID 2001:3319 D-Link Corp. Wireless N Nano USB Adapter```

-----
1 - I am your Repo on all PCs rtl8192eu installed 2 years back, issue was there from that time.

vijaytejak, 2021-11-26:
Hi @CGarces is the above info correct ?