Mange/rtl8192eu-linux-driver #260: DKMS giving a headache at install

Shumy52, 2021-12-01:
I've installed it a thousand times because I reinstalled linux a thousand times. But this time when running the command:

"sudo dkms add . "

from step 4 on that installation manual, it gives out this error:

"Error! Arguments <module> and <module-version> are not specified."

I also tried to mention the whole location of where the git clone was but the error persits. I have some good reasons to believe that I didn't miss the file that I was supposed to use DKMS on.

It worked earlier today, I'm trying to do a fresh install of Manjaro and I need the driver for my WI-FI adapter. Today I've also tried to install EndeavourOS and Arco linux just for the hell of it but this command worked. At this point my brain is melting... Am I missing something? Was there an update that I missed... Thank you for your time.

TuxStory, 2021-12-10:
Hello

I compiled & installed the driver 
Device-2: TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU] type: USB driver: rtl8192eu

But I got the same issue with DKMS : 

```
[@Lenovo rtl8192eu-linux-driver]$ sudo dkms add .
[sudo] Mot de passe de : 
Error! Arguments <module> and <module-version> are not specified.
Usage: add <module>/<module-version> or
       add -m <module>/<module-version> or
       add -m <module> -v <module-version>
```

Linux Lenovo 5.15.2-2-MANJARO #1 SMP PREEMPT Sat Nov 13 19:25:38 UTC 2021 x86_64 GNU/Linux

Shumy52, 2021-12-13:
Unfortunately I gave up and switched to POP OS and there it worked. I don't know what is wrong on Manjaro, still haven't found a solution