Mange/rtl8192eu-linux-driver #261: net/ipx.h: No such file or directory

Peterdes, 2022-01-06:
Remove IPX support from driver, set as obsolete in Jan 2018.
IPX is not supported by the Linux kernel since v5.15-rc1, see commit 6c9b40844751ea30c72f7a2f92f4d704bc6b2927

Trying to compile with 5.15.12 kernel yields the error:
`/var/lib/dkms/rtl8192eu/1.0/build/core/rtw_br_ext.c:20:18: fatal error: net/ipx.h: No such file or directory`

Peterdes, 2022-01-06:
I guess something like that should work:
https://github.com/aircrack-ng/rtl8188eus/pull/146/commits/4830d3906230a4d80ba67709a06c9d5b99764839

CGarces, 2022-01-06:
Hi.
Compilation works fine on 5.15 since Octobrer 2021
https://github.com/Mange/rtl8192eu-linux-driver/runs/3988904696?check_suite_focus=true
See #244 

Are you using a custom kernel from CentOS?

Peterdes, 2022-01-06:
Oh my. I just I haven't updated the code to the latest version... sorry for the fuss.

closed 2022-01-06 by Peterdes