Mange/rtl8192eu-linux-driver #262: Driver doesn't load

KekOnTheWorld, 2022-01-30:
First of all, thank you for sharing this. I have an old laptop with a Intel pentium 3 (x86 no sse2 support) and installed debian 11 (linux-image-5.10.0-10-686). I bought a wifi adapter (TP-Link TL-WN823N v2/v3 [Realtek RTL8192EU]) which required this driver because the driver on the official tplink page was too outdated for my kernel version.  I followed all the necessary steps on here. After i rebooted i typed sudo lshw -c network but there was just my ethernet card built in.

Full stdout of sudo lshw -c network:
```
*-network
  description: Ethernet interface
  product: 82801CAM (ICH3) PRO/100 VE (LOW) Ethernet Controller
  vendor: Intel Corporation
  physical id: 8
  bus info: pci@0000:02:08.0
  logical name: enp2s8
  version: 41
  serial: 00:d0:59:cd:81:63
  size: 100 Mbit/s
  capacity: 100 Mbit/s
  width: 32 bits
  clock: 33MHz
  capabilities: pm bus_master cap_list ethernet physical tp mii 10bt 10bt-fd autonegotiation
  configuration: autonegotiation=on broadcast=yes driver=e100 driverversion=5.10.0-11-686 duplex=full ip=192.168.178.71 latency=66 link=yes maxlatency=56 mingnt=8 multicast=yes port =MII speed=100Mbit/s
  resources: irq:11 memory:c0200000-c0200fff ioport:6400(size=64)
```

What was a bit weird was that the stdout of sudo update-initramfs returned following:
```
update-initramfs: Generating /boot/initrd.img-5.10.0-11-686
W: Possible missing firmware /lib/firmware/e100/d102e_ucode.bin for module e100
W: Possible missing firmware /lib/firmware/e100/d101s_ucode.bin for module e100
W: Possible missing firmware /lib/firmware/e100/d101m_ucode.bin for module e100
```

KekOnTheWorld, 2022-01-30:
lsmod doesnt contain this driver neither.

KekOnTheWorld, 2022-01-30:
stdout of sudo dkms status:
```
rtl8192eu, 1.0, 5.10.0-10-686, i686: installed
```
It seams like it is installed on the wrong kernel version?!?

KekOnTheWorld, 2022-01-30:
Okay i rebuilt the driver and now everything works

closed 2022-01-30 by KekOnTheWorld

AndreiCherniaev, 2022-11-15:
**firmware d102e_ucode.bin for driver e100 missing**

How I can fix it? Method A. Do steps:
0) My board's driver mount to my laptop on _/media/a/rootfs_
1) cd /media/a/rootfs/lib/firmware/
2) mkdir e100 
3) copy [d102e_ucode.bin](https://gitlab.freedesktop.org/frankbinns/linux-firmware/-/tree/powervr/e100) to your board's drive
sudo cp /home/a/myb/d102e_ucode.bin /media/a/rootfs/lib/firmware/
4) Now it should be work.

Or use alternative method B. Just set this:

[image]