Mange/rtl8192eu-linux-driver #268: running the driver on OpenWrt 21.02.2

artynet, 2022-04-13:
Hello @CGarces,

I've recently embarked in the porting of the rtl8192eu driver for the OpenWrt embedded linux systems. Starting from the only Realtek officially supported driver here:

https://github.com/openwrt/openwrt/tree/master/package/kernel/rtl8812au-ct

I have applied similar patches to the ongoing development branch of the driver **realtek-4.4.x**:

https://github.com/artynet/rtl8192eu-linux-driver/tree/openwrt

then, accordingly the Makefile has been written to properly build the driver:

```makefile
#
# Copyright (C) 2006-2009 OpenWrt.org
#
# This is free software, licensed under the GNU General Public License v2.
# See /LICENSE for more information.
#

include $(TOPDIR)/rules.mk

PKG_NAME:=rtl8192eu-ct-mange
PKG_RELEASE=2

PKG_LICENSE:=GPLv2
PKG_LICENSE_FILES:=

PKG_SOURCE_URL:=https://github.com/Mange/rtl8192eu-linux-driver.git
# PKG_MIRROR_HASH:=fa689e034cad9e4683ea784b8f3cb590492ab5c68e8babd492a4e8bf2de3b114
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2022-04-06
PKG_SOURCE_VERSION:=5484fbd00c39826fd91e2d784cce211d55537447

PKG_MAINTAINER:=mantainer <mantainer@github.com>
PKG_BUILD_PARALLEL:=1
#PKG_EXTMOD_SUBDIRS:=rtl8192eu-ct-mange

STAMP_CONFIGURED_DEPENDS := $(STAGING_DIR)/usr/include/mac80211-backport/backport/autoconf.h

include $(INCLUDE_DIR)/kernel.mk
include $(INCLUDE_DIR)/package.mk

define KernelPackage/rtl8192eu-ct-mange
  SUBMENU:=Wireless Drivers
  TITLE:=Driver for Realtek 8192 EU devices comfast 912-ac, etc
  DEPENDS:=+kmod-cfg80211 +kmod-usb-core +@DRIVER_11N_SUPPORT +@DRIVER_11AC_SUPPORT
  FILES:=\
	$(PKG_BUILD_DIR)/8192eu.ko
  AUTOLOAD:=$(call AutoProbe,8192eu)
  PROVIDES:=kmod-rtl8192eu-mange
endef

NOSTDINC_FLAGS = \
        $(KERNEL_NOSTDINC_FLAGS) \
        -I$(PKG_BUILD_DIR) \
        -I$(STAGING_DIR)/usr/include/mac80211-backport/uapi \
        -I$(STAGING_DIR)/usr/include/mac80211-backport \
        -I$(STAGING_DIR)/usr/include/mac80211/uapi \
        -include backport/backport.h \
        -I$(PKG_BUILD_DIR)/include \
        -I$(STAGING_DIR)/usr/include/mac80211

NOSTDINC_FLAGS+= -DCONFIG_IOCTL_CFG80211 -DRTW_USE_CFG80211_STA_EVENT -DBUILD_OPENWRT

ifdef CONFIG_PACKAGE_kmod-rtl8192eu-ct-mange
   PKG_MAKE_FLAGS += CONFIG_RTL8192EU=m
   PKG_MAKE_FLAGS += CONFIG_RTW_LOG_LEVEL=1
   PKG_MAKE_FLAGS += CONFIG_POWER_SAVING="n"
   PKG_MAKE_FLAGS += CONFIG_PLATFORM_OPENWRT_LITTLE_ENDIAN="y"
   PKG_MAKE_FLAGS += CONFIG_PLATFORM_ARM_RPI="y"
   PKG_MAKE_FLAGS += CONFIG_WIFI_MONITOR="y"
endif

define Build/Compile
	+$(MAKE) $(PKG_JOBS) -C "$(LINUX_DIR)" \
		$(KERNEL_MAKE_FLAGS) \
		$(PKG_MAKE_FLAGS) \
		USER_EXTRA_CFLAGS="$(EXTRA_CFLAGS)" \
		M="$(PKG_BUILD_DIR)" \
		NOSTDINC_FLAGS="$(NOSTDINC_FLAGS)" \
		modules
endef

$(eval $(call KernelPackage,rtl8192eu-ct-mange))
```
so far, the driver builds properly so to allow compatibility with the internal **cfg802.11/mac802.11** OpenWrt stack. Once the system is booted, the driver is properly loaded but unfortunately when running one of the standard commands to show the active network interfaces `ifconfig` or `ip a` the system freezes and the only option to unlock it is to perform a general reboot.

Can you give some pointers about that ? I browsed through the closed issues and time ago a couple of developers asked for your help but in the very end they turned to build (maybe) an official driver release for a specific USB dongle....

If I could figure out the issue maybe it would open the doors for the porting of other realtek drivers to this very embedded operating system. Thanks in advance. I will appreciate you support....

Kind Regards

CGarces, 2022-04-13:
Sorry, I can't help with that.
Your first option must be use mainline kernel, try to compile the kernel with CONFIG_RTL8XXXU=m and CONFIG_RTL8XXXU_UNTESTED=y

Looks like your issue is common in rtl drivers
https://github.com/aircrack-ng/rtl8812au/issues/514

Take a look at https://github.com/aircrack-ng/rtl8812au issues and commits, maybe you can found something interesting.

artynet, 2022-04-13:
Thank you @CGarces,

I thoroughly browsed those discussions time ago end ended up looking here:

https://github.com/abperiasamy/rtl8812AU_8821AU_linux/pull/285

alas, the **8192eu** driver structure is very different from the **8812au** one altough they share a lot in common. This set of patches can't be easily reworked to fit this driver so I think I have to dig deeper into the source code...If I ever come up with a solution I'll share it with the whole community of course.

Please don't close soon the issue so to have other feedback about it....

Regards, Arturo

xirixiz, 2023-12-29:
Hi @artynet, I was exploring the same thing. Were you able to find/fix this? If not, I`ll give up as well :)