Mange/rtl8192eu-linux-driver #270: Updates for 5.17

bwduncan, 2022-05-09:
Hi, since recently rtl8192eu has failed to compile on the latest upstream kernel. I'm currently using 5.17.6 because there are suspend fixes for my machine that are not in the Ubuntu kernels yet. The error is:

```
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘thread_exit’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:878:9: error: implicit declaration of function ‘complete_and_exit’ [-Werror=implicit-function-declaration]
  878 |         complete_and_exit(comp, 0);
      |         ^~~~~~~~~~~~~~~~~
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c: In function ‘rtw_change_ifname’:
/var/lib/dkms/rtl8192eu/1.0/build/os_dep/osdep_service.c:1589:23: warning: passing argument 1 of ‘memcpy’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
 1589 |         memcpy(pnetdev->dev_addr, adapter_mac_addr(padapter), ETH_ALEN);
      |                ~~~~~~~^~~~~~~~~~
```

It looks like `complete_and_exit` has been renamed `kthread_complete_and_exit` by cead18552660702a4a46f58e65188fe5f36e9dfe.

This commit (from a different project!) might be relevant: https://github.com/cilynx/rtl88x2bu/commit/8bbaa55852ed0c4f3d6370acde17551d31125361

CGarces, 2022-05-09:
Hi.

This issue was fixed on
https://github.com/Mange/rtl8192eu-linux-driver/commit/5fb30bc593d5739a0942ebd33278e4e4f831fd3e
And compilation for 5.17 is working fine
https://github.com/Mange/rtl8192eu-linux-driver/runs/5850581061?check_suite_focus=true

The thread_exit function was simplified on https://github.com/Mange/rtl8192eu-linux-driver/commit/352330be93bfcc4e8071b8fc3192c8a8321d64dc

Are you sure that are using the last version of branch **realtek-4.4.x**?

Please take a look at the automated builds
https://github.com/Mange/rtl8192eu-linux-driver/actions

closed 2022-05-09 by CGarces

bwduncan, 2022-05-09:
Oh yeah thanks you are right I had not updated the dkms package. Sorry!